ni/nimi-python #806: Add NI-SCOPE Measurement Library support

Fladolcetta, 2018-04-12:
All of the measurement library functions are not included because as they are, they would be pythonic.

We should add support for them with pythonic wrappers

charterchap, 2018-08-20:
Really wanting this one. I'd rather have something that works, than have something "pythonic". I'm looking at http://zone.ni.com/reference/en-XX/help/370592AB-01/digitizers/fetching_scalar_and_array_measurements/ and not seeing a way to do it with nimi-python :(

Fladolcetta, 2018-08-21:
@charterchap I believe that we marked the measurement library methods private. If you needed measurment library at this time, you might be able to try this out. That being said when we officially support measurement library, your code might break.

Where we mark it as private:
https://github.com/ni/nimi-python/blob/1f193aa260ead422188b8f7a81929bd9c12cce60/src/niscope/metadata/functions_addon.py#L63-L71

The generated file where one of these private functions are defined:
https://github.com/ni/nimi-python/blob/1f193aa260ead422188b8f7a81929bd9c12cce60/generated/niscope/session.py#L1609

charterchap, 2018-08-21:
It is working more or less. I had to edit session.py in the niscope module on line 2835
```
if type(scalar_meas_function) is not enums.ScalarMeasurement:
            raise TypeError('Parameter mode must be of type ' + str(enums.ScalarMeasurement))
```
to
```
if type(scalar_meas_function) is not enums._ScalarMeasurement:
            raise TypeError('Parameter mode must be of type ' + str(enums._ScalarMeasurement))
```

a quick example for anyone else who comes across this:
`print(session.channels[channels]._fetch_measurement(niscope.enums._ScalarMeasurement.FREQUENCY))`

marcoskirsch, 2018-08-21:
@texasaggie97 The replacement @charterchap had to do sounds like a small generator bug. I've created #922.

PR #1485 (2020-08-05): NI-SCOPE: Add measurement library methods
- [x] This contribution adheres to [CONTRIBUTING.md](https://github.com/ni/nimi-python/blob/master/CONTRIBUTING.md).

- [x] I've updated [CHANGELOG.md](https://github.com/ni/nimi-python/blob/master/CHANGELOG.md) if applicable.

- [x] I've added tests applicable for this pull request

### What does this Pull Request accomplish?

This PR adds public-facing wrappers for two private methods in the measurement library, fetch_measurement_stats and fetch_array_measurement. The custom type MeasurementStats is added to NI-SCOPE for returning data in a pythonic  [... cut]
closed 2020-08-05 by sbethur