ni/nimi-python #811: Parameters that use converter require duplicate metadata

marcoskirsch, 2018-04-17:
Observed in https://github.com/ni/nimi-python/pull/803#pullrequestreview-112158144
When adding support for `floats in seconds` or `timedelta`, two pieces of metadata were specified over and over: 

``` python
{ 'python_api_converter_name': 'convert_timedelta_to_milliseconds',
'type_in_documentation': 'float in seconds or datetime.timedelta',}
```

The truth is that anything using any of the `convert_timedelta_to_*` functions would have the same exact `type_in_documentation`. This is only room for maintenance errors. It would be better if we could somehow associate converter functions to type_in_documentation outside of metadata so that in metadata only one thing needs to be specified. 

I have no suggestions currently on how to best do this, especially because we need the info at code generation time.

marcoskirsch, 2018-04-20:
Another instance of this:

``` python
'GetExtCalRecommendedInterval':     { 'parameters': { 1: { 'python_api_converter_name': 'convert_month_to_timedelta',  'type_in_documentation': 'datetime.timedelta', }, }, },
```

We know that whenever `convert_month_to_timedelta` is used, `'type_in_documentation'` will always be `'datetime.timedelta'`.

Yet we copied it 4 times in #813.

MrDuff8, 2019-08-01:
This information is now provided by NI internal tools. We'll track on our own internal backlog.

closed 2019-08-01 by MrDuff8