ni/nimi-python #818: Configure tox interpreter search error

hoffiz, 2018-04-23:
While following the instructions to build the API following the next instructions (non WSL route):
```
"C:\Program Files\Python36\python" -m tox --scan C:/Program*/Python*/python.exe
```
errors with:
``` 
Traceback (most recent call last):
  File "C:\Program Files\Python36\lib\runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "C:\Program Files\Python36\lib\runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "C:\Program Files\Python36\lib\site-packages\tox\__main__.py", line 4, in <module>
    run_main()
  File "C:\Program Files\Python36\lib\site-packages\tox\session.py", line 40, in run_main
    main(args)
  File "C:\Program Files\Python36\lib\site-packages\tox\session.py", line 45, in main
    config = prepare(args)
  File "C:\Program Files\Python36\lib\site-packages\tox\session.py", line 27, in prepare
    config = parseconfig(args)
  File "C:\Program Files\Python36\lib\site-packages\tox\config.py", line 278, in parseconfig
    pm.hook.tox_configure(config=config)
  File "C:\Program Files\Python36\lib\site-packages\pluggy\__init__.py", line 617, in __call__
    return self._hookexec(self, self._nonwrappers + self._wrappers, kwargs)
  File "C:\Program Files\Python36\lib\site-packages\pluggy\__init__.py", line 222, in _hookexec
    return self._inner_hookexec(hook, methods, kwargs)
  File "C:\Program Files\Python36\lib\site-packages\pluggy\__init__.py", line 216, in <lambda>
    firstresult=hook.spec_opts.get('firstresult'),
  File "C:\Program Files\Python36\lib\site-packages\pluggy\callers.py", line 201, in _multicall
    return outcome.get_result()
  File "C:\Program Files\Python36\lib\site-packages\pluggy\callers.py", line 76, in get_result
    raise ex[1].with_traceback(ex[2])
  File "C:\Program Files\Python36\lib\site-packages\pluggy\callers.py", line 180, in _multicall
    res = hook_impl.function(*args)
  File "C:\Program Files\Python36\lib\site-packages\tox_globinterpreter.py", line 314, in tox_configure
    return global_interpreter_plugin.tox_configure(config)
  File "C:\Program Files\Python36\lib\site-packages\tox_globinterpreter.py", line 237, in tox_configure
    self.scan()
  File "C:\Program Files\Python36\lib\site-packages\tox_globinterpreter.py", line 255, in scan
    il.write(file_name)
  File "C:\Program Files\Python36\lib\site-packages\tox_globinterpreter.py", line 129, in write
    assert ' ' not in base_name
TypeError: a bytes-like object is required, not 'str'
```
but I can successfully run the command with Python27
```
"C:\Program Files\Python27\python" -m tox --scan "C:/Program*/Python*/python.exe"
interpreters:
python2.7 c:\Program Files\Python27\python.exe
python3.6 c:\Program Files\Python36\python.exe
python2.7 c:\Program Files (x86)\Python27\python.exe
python3.6 c:\Program Files (x86)\Python36\python.exe
```
I understand this is outside the scope of nimi-python but was curious if it is ok to collect the interpreters with Python27 or if there is any suggestion of what could be happening?

texasaggie97, 2018-04-30:
Using Python 2.7 is fine.

`tox_globinterpreter`'s --scan option doesn't seem to work with Python 3.x. The instructions use Python 2 for the --scan command, but don't specifically call it out to make it obvious. Should I add something to [CONTRIBUTING.md](https://github.com/ni/nimi-python/blob/master/CONTRIBUTING.md) to make it more obvious?

texasaggie97, 2018-10-19:
[CONTRIBUTING.md](https://github.com/ni/nimi-python/blob/master/CONTRIBUTING.md) now explicitly says to use Python 2.

Closing this issue.

PR #944 (2018-10-30): Correct and add more to contributing
- [X] This contribution adheres to [CONTRIBUTING.md](https://github.com/ni/nimi-python/blob/master/CONTRIBUTING.md).
- [X] ~~I've updated [CHANGELOG.md](https://github.com/ni/nimi-python/blob/master/CHANGELOG.md) if applicable.~~
- [X] ~~I've added tests applicable for this pull request~~
### What does this Pull Request accomplish?
* Small edits to clarify that we need python 2
* Should have been part of closing #818 

### List issues fixed by this Pull Request below, if any.
* Fix #818 

### What testing has been done?
* Visual inspection of `CONTRIBUTING. [... cut]
closed 2018-10-19 by texasaggie97