ni/nimi-python #855: Does fetch_into method does not support the acqusition using 2 channels?

swapnil15689, 2018-05-14:
Hello,
I am using a PXI 5922 niscope with 8MB onboard memory per channel. I want to read the continuous samples from two channels for a longer time (so only one record but very long); so I use the fetch_into method.

My code works perfectly fine for single channel for any acquisition freq  but no matter what, it does not work for 2 channels and I end up with the error saying 'The requested data has been overwritten in memory so it is no longer available for fetching.'
The code follows:

scope = niscope.Session("Dev1")
scope.configure_horizontal_timing(50000,131072,0.0,1,True)
scope.channels['0,1'].configure_vertical(5.0, coupling=niscope.VerticalCoupling.DC,enabled=True)
scope.allow_more_records_than_memory=True
scope.initiate()
wfm=np.ndarray((4194304),dtype=np.float64)
wfm_info=scope.channels['0,1'].fetch_into(wfm,relative_to=FetchRelativeTo.READ_POINTER,offset=0,record_number=0, num_records=1,timeout=datetime.timedelta(seconds=-1.0))
scope.close()

The above code works perfectly fine for single channel. Is there something which I am missing..?

Thanks,
Swapnil

marcoskirsch, 2018-05-14:
* What version of the driver runtime are you using?
* What version of niscope Python module are you using?
* What kind of system are you running this on?

Thanks

swapnil15689, 2018-05-14:
I am using:
runtime 16.0
niscope python 0.8.0
Chassis: NI PXle-1071; Controller: NI PXle-8820; Windows 7 64 bits

Thanks

marcoskirsch, 2018-05-15:
Please try two things:

1. Swap the line in which you create the `np.ndarray` with the line in which you `initiate()`. This way the device is not acquiring while the memory is getting allocated, wasting precious on-board memory.
2. Try decreasing your sample rate to half or even less if needed. Your program may work with one channel, but maybe your system (2.2 GHz Intel Celeron 1020E) may truly not be able to keep up with twice the amount of data. If that's the case, then we have ruled out bugs and we can look into other ways to modify the program so that it can keep up.

swapnil15689, 2018-05-15:
1. I tried the first approach you suggested; but it did not work; I got the same error.

2. In fact, 50kHz is the minimum acquisition frequency for PXI 5922. For single channel it works fine for 50kHz and also even for 15MHz which is the maximum limit for acquisition frequency with PXI 5922.

Thanks

Fladolcetta, 2018-05-15:
We are currently looking into this right now. I had some additional questions: 

1. What version of Python are you using?
2. I know that you said you were running Windows 7 64-bit, but are you also running Python 64-bit as well?

swapnil15689, 2018-05-16:
1. I am using Python 2.7.14 
2. Yes, I am running 64-bit python as well

Thanks

texasaggie97, 2018-05-17:
Hi @swapnil15689,
What are you trying to do?
* Do you want to acquire many records of 128K samples?
* Do you want to acquire a single record which is > 8MB of memory per channel?
* Something else?

Keep in mind that the PXI-5922 is flexible resolution so the slower you go, the more memory per sample you will need. I believe at this rate, it uses 4 bytes per sample.

Once I know which you are trying to accomplish, I can better determine what is going wrong in the script.

swapnil15689, 2018-05-18:
Hi,

1. I want to acquire only one record . In that one record, I want to acquire samples which are always more than 2097052(2097052 samples is the limit of PXI 5922).
2. Yes, I want to acquire a single record for which I need more than the onboard memory; in this case >8MB per channel
3. Typically for my applications I need to acquire data using 1MHz or more acquisition frequency for 100-120 seconds(using both channels at the same time)

Thanks 

-

texasaggie97, 2018-05-30:
It does work. Here is a portion of an example I am trying to get merged in PR #880 
``` python
import argparse
import datetime
import niscope
import numpy as np
import pprint
import sys

pp = pprint.PrettyPrinter(indent=4, width=80)

# This example is based on the C FetchForever example
def example(resource_name, channels, options, total_acquisition_time_in_seconds, voltage, sample_rate_in_hz, samples_per_fetch):
    total_samples = int(total_acquisition_time_in_seconds * sample_rate_in_hz)
    channel_list = channels.split(',')  # We need channels as a list

    # 1. Creating numpy arrays
    waveforms = [np.ndarray(total_samples, dtype=np.float64) for c in channel_list]

    # 2. Opening session
    with niscope.Session(resource_name) as session:
        # 3. Configuring
        session.configure_horizontal_timing(min_sample_rate=sample_rate_in_hz, min_num_pts=1, ref_position=0.0, num_records=1, enforce_realtime=True)
        session.channels[channel_list].configure_vertical(voltage, coupling=niscope.VerticalCoupling.DC, enabled=True)
        # We configure for a software trigger and then never send it
        session.configure_trigger_software()
        current_pos = 0
        # 4. initating
        with session.initiate():
            while current_pos < total_samples:
                # We fetch each channel at a time so we don't have to de-interleave afterwards
                # We do not keep the wfm_info returned from fetch_into
                for c, wfm in zip(channel_list, waveforms):
                    # 5. fetching
                    session.channels[c].fetch_into(wfm[current_pos:current_pos + samples_per_fetch], relative_to=niscope.FetchRelativeTo.READ_POINTER,
                                                   offset=0, record_number=0, num_records=1, timeout=datetime.timedelta(seconds=5.0))
                current_pos += samples_per_fetch

        # 6 fetch complete
    # 7 session closed
```

closed 2018-05-30 by texasaggie97