ni/nimi-python #865: Remove offset/gain/amplitude from nifgen functions

marcoskirsch, 2018-05-17:
> Explicit is better than implicit.
[Zen](https://www.python.org/dev/peps/pep-0020/#id3)

* `configure_freq_list(self, frequency_list_handle, amplitude, dc_offset=0.0, start_phase=0.0)`
* `configure_standard_waveform(self, waveform, amplitude, frequency, dc_offset=0.0, start_phase=0.0)`

Look for other instances.

texasaggie97, 2018-05-31:
I am not sure what this means. Do you mean remove default values?

marcoskirsch, 2018-06-01:
> I am not sure what this means. Do you mean remove default values?

Yes, fixed title.

texasaggie97, 2018-06-04:
How do we determine which parameters get a default value? Should `start_phase` for instance?

texasaggie97, 2018-06-05:
If we go by
> Explicit is better than implicit.

then nothing would have defaults. I don't think that would be the right thing to do. In the examples you have, I think those defaults are appropriate.

texasaggie97, 2018-06-05:
The only parameters' default value I think might be questionable are:
* `level` on `configure_trigger_edge()` - default value == 0.0
* `level` on `configure_trigger_hysteresis()` - default value == 0.0
* `hysteresis` on `configure_trigger_hysteresis()` - default value == 0.05

marcoskirsch, 2018-06-05:
ok

PR #890 (2018-06-05): Remove a few default values from niscope
- [X] This contribution adheres to [CONTRIBUTING.md](https://github.com/ni/nimi-python/blob/master/CONTRIBUTING.md).
- [X] I've updated [CHANGELOG.md](https://github.com/ni/nimi-python/blob/master/CHANGELOG.md) if applicable.
- [X] I've ~~added~~updated tests applicable for this pull request
### What does this Pull Request accomplish?
* Removed default value for `level` parameter on `configure_trigger_edge()`
    * paramater list is now
        ``` python
        configure_trigger_edge(self, trigger_source, level, trigger_coupling, slope=enums.TriggerSl [... cut]
closed 2018-06-05 by texasaggie97