ni/nimi-python #881: niswitch review notes

marcoskirsch, 2018-05-30:
# Description of issue

We reviewed the niswitch module with NI-SWITCH domain experts. We have a few action items prior version 1.0.

Remove the following attributes that apply only to obsolete hardware

* cabled_module_scan_advanced_bus = _attributes.AttributeViInt32(1150009)
* cabled_module_trigger_bus = _attributes.AttributeViInt32(1150008)
* master_slave_scan_advanced_bus = _attributes.AttributeViInt32(1150007)
* master_slave_trigger_bus = _attributes.AttributeViInt32(1150006)
* parsed_scan_list = _attributes.AttributeViString(1150012)
* trigger_mode = _attributes.AttributeViInt32(1150005)

Remove the following methods, since they have corresponding properties

* configure_scan_list(self, scanlist, scan_mode=enums.ScanMode.BREAK_BEFORE_MAKE)
* configure_scan_trigger(self, trigger_input, scan_advanced_output, scan_delay=datetime.timedelta(seconds=0.0))
* route_scan_advanced_output(self, scan_advanced_output_connector, scan_advanced_output_bus_line, invert=False)
* route_trigger_input(self, trigger_input_connector, trigger_input_bus_line, invert=False)
* set_continuous_scan(self, continuous_scan)

Be consistent about trigger polarity. Did we remove in `nifgen` and `nidcpower`?
* scan_advanced_polarity = _attributes.AttributeEnum(_attributes.AttributeViInt32, enums.ScanAdvancedPolarity, 1150011)
* trigger_input_polarity = _attributes.AttributeEnum(_attributes.AttributeViInt32, enums.TriggerInputPolarity, 1150010)

This property is missing an enum
* wire_mode = _attributes.AttributeViInt32(1250017)

Private initiate referenced in docs. This needs to be fixed in metadata.

texasaggie97, 2018-05-31:
Where is the enum for `wire_mode` defined? I do not see one in either CVI nor LabVIEW.

texasaggie97, 2018-05-31:
I am trying to remove `route_scan_advanced_output()`. I can only find one attribute that uses enum `ScanAdvancedOutput` but that function takes two, parameter `scan_advanced_output_connector` and scan_advanced_output_bus_line`.
What attributes correspond to those two parameters?

texasaggie97, 2018-05-31:
While we removed polarity attributes in `nifgen` and `nidcpower`, we left them for measurement complete in `nidmm`. Since that seems like the most likely use for `scan_advanced_polarity` and `trigger_input_polarity` it seems like we should leave it for `niswitch` as well.

Fladolcetta, 2018-05-31:
> I am trying to remove `route_scan_advanced_output().` I can only find one attribute that uses enum `ScanAdvancedOutpu`t but that function takes two, parameter `scan_advanced_output_connector` and `scan_advanced_output_bus_line`.
> What attributes correspond to those two parameters?

I am not sure if there is a specific attribute to do that. This function basically takes TTL0 and routes it to a front connector on some switches. The inverse is true for `route_trigger_input()`. This does some internal routing I believe. 

I personally believe that the likelihood of someone using these functions are low. We can maybe just make it private?

Fladolcetta, 2018-05-31:
> Where is the enum for `wire_mode` defined? I do not see one in either CVI nor LabVIEW.

There is NOT an enum for LabVIEW. The driver returns an int for "1", "2," or "4". Perhaps we leave it as an int, or create the enum in enums_addon.py?

marcoskirsch, 2018-05-31:
> Perhaps we leave it as an int, or create the enum in enums_addon.py?

Leave as int. We removed enums that are simple numeric constants elsewhere (i.e. power line frequency in NI-DMM).
I incorrectly assumed it used an enum in other languages.

texasaggie97, 2018-05-31:
> > I am trying to remove `route_scan_advanced_output()`. I can only find one attribute that uses enum `ScanAdvancedOutput` but that function takes two, parameter `scan_advanced_output_connector` and `scan_advanced_output_bus_line.`
What attributes correspond to those two parameters?

> I am not sure if there is a specific attribute to do that. This function basically takes TTL0 and routes it to a front connector on some switches. The inverse is true for route_trigger_input(). This does some internal routing I believe.

> I personally believe that the likelihood of someone using these functions are low. We can maybe just make it private?

I am leaving it as is. If there are not attributes that can be used, I don't think we can remove it.

PR #886 (2018-06-01): Changes from niswitch review
- [X] This contribution adheres to [CONTRIBUTING.md](https://github.com/ni/nimi-python/blob/master/CONTRIBUTING.md).
- [X] I've updated [CHANGELOG.md](https://github.com/ni/nimi-python/blob/master/CHANGELOG.md) if applicable.
- [X] ~~I've added tests applicable for this pull request~~
### What does this Pull Request accomplish?
* Changes from niswitch review
  * Remove attributes that only apply to EOL HW
    * `cabled_module_scan_advanced_bus`
    * `cabled_module_trigger_bus`
    * `master_slave_scan_advanced_bus`
    * `master_slave_trigger_bus`
    * `parsed_sc [... cut]
closed 2018-06-01 by marcoskirsch