ni/nimi-python #887: Installation instructions are wrong in documentation

marcoskirsch, 2018-06-01:
* Description of issue

Looking at [installation instructions](http://nimi-python.readthedocs.io/en/master/niscope.html#installation) for `niscope`:

```
python -m pip install niscope~=1.0.0.dev0

Collecting niscope~=1.0.0.dev0
  Could not find a version that satisfies the requirement niscope~=1.0.0.dev0 (from versions: 0.5.0, 0.6.0, 0.7.0, 0.8.0.dev0, 0.8.0, 0.9.0)
No matching distribution found for niscope~=1.0.0.dev0
```

1.0.0 is not yet released. But I wonder if this will be a maintenance burden that we should address.

texasaggie97, 2018-06-06:
This only applies to master. The released readme's are correct. For an example, look at the 0.9.0 readme that is on [PyPI](https://pypi.org/project/nidmm/)

[image]

Do we care about this in master?

Fladolcetta, 2018-06-07:
Why list the specific version at all?

Fladolcetta, 2018-06-07:
Closing because Master documentation is pointing to the version of the master trunk that is not released. The documentation on PyPI is correct for the released version.

closed 2018-06-07 by Fladolcetta