ni/nimi-python #895: TypeError: Invalid first argument to register()

charterchap, 2018-07-02:
* Description of issue
Using Python 3.7.0,  Can not import niswitch (0.9.0 and 1.0.0)

    ```
=================================== ERRORS ====================================
__________________ ERROR collecting load_default_settings.py __________________
load_default_settings.py:10: in <module>
    from session import Session, Consts
..\lib\session.py:13: in <module>
    from pgvts_power import *
..\lib\pgvts_power.py:9: in <module>
    from switch_manager import SwitchManager
..\lib\switch_manager.py:9: in <module>
    import niswitch
C:\Program Files (x86)\Python37-32\lib\site-packages\niswitch\init.py:7: in <module>
    from niswitch.session import Session  # noqa: F401
C:\Program Files (x86)\Python37-32\lib\site-packages\niswitch\session.py:7: in <module>
    import niswitch._attributes as _attributes
C:\Program Files (x86)\Python37-32\lib\site-packages\niswitch\_attributes.py:3: in <module>
    import niswitch._converters as _converters
C:\Program Files (x86)\Python37-32\lib\site-packages\niswitch\_converters.py:54: in <module>
    @_convert_repeated_capabilities.register(six.string_types)  # noqa: F811
C:\Program Files (x86)\Python37-32\lib\functools.py:801: in register
    f"Invalid first argument to `register()`: {cls!r}. "
E   TypeError: Invalid first argument to `register()`: (<class 'str'>,). Use either `@register(some_class)` or plain `@register` on an annotated function.
!!!!!!!!!!!!!!!!!!! Interrupted: 1 errors during collection !!!!!!!!!!!!!!!!!!!
    ```

* Steps to reproduce issue

    ``` python
    import niswitch
    ```

This may be of interest:
https://github.com/python/cpython/commit/e56975351bc2c574c728f738e88effba7116279f

marcoskirsch, 2018-07-03:
1. Do you have numpy? I'm wondering if it is related to https://github.com/ni/nimi-python/issues/894
2. You mention `niswitch` only. Is that because that's the only module from nimi-python that you tried, or because it is the only one that broke? I'm wondering if it's something specific to `niswitch` or if the issue affects all modules.

charterchap, 2018-07-09:
It does work fine with python 3.6.5. I do have numpy installed.  niswitch is the first of the nimi-python imports, so i'm not sure if the others have the same problem (i've since downgraded python).
Take a look at the commit in the python code base from my original post. I think they changed something on you with @register.

charterchap, 2018-07-09:
https://github.com/python/cpython/pull/4733

PR #917 (2018-08-20): Support Python 3.7
- [X] This contribution adheres to [CONTRIBUTING.md](https://github.com/ni/nimi-python/blob/master/CONTRIBUTING.md).
- [X] I've updated [CHANGELOG.md](https://github.com/ni/nimi-python/blob/master/CHANGELOG.md) if applicable.
- [X] ~~I've added tests applicable for this pull request~~
### What does this Pull Request accomplish?
* Fix issues with Python 3.7
* Add official support for Python 3.7

### List issues fixed by this Pull Request below, if any.
* Fix #895 

### What testing has been done?
* Unit
* System
closed 2018-08-20 by marcoskirsch