ni/nimi-python #902: nidcpower missing methods for configure_digital_edge_measure_trigger and export_signal

user0129, 2018-07-10:
I have been using nidcpower version 0.8.0 previously (as well as the c dll in the past) and the following methods were available in the session.py:
"configure_digital_edge_measure_trigger" and "export_signal". 
However, having recently switched to version 1.0.0, these methods seem to have been removed. Is there a replacement for these functions? Or any documentation that addresses this issue where I can get information on what to use instead?

Thanks.

marcoskirsch, 2018-07-10:
Hello.

Per our own [API design guidelines](https://github.com/ni/nimi-python/wiki/API-design-guidelines) and [Zen of Python](https://www.python.org/dev/peps/pep-0020/#id3), we removed redundant configuration options. Turns out that the functions you are asking about are nothing more than wrappers around properties.

* Changes were made in #872 
* [CHANGELOG](https://github.com/ni/nimi-python/blob/master/CHANGELOG.md) enumerates the changes.
* You can see documentation for all properties in our [NI Modular Instruments Python Documentation](http://nimi-python.readthedocs.io/en/1.0.0/).

Use:
```
nidcpower.Session.digital_edge_measure_trigger_input_terminal
```
instead of:
```
nidcpower.Session.configure_digital_edge_measure_trigger(input_terminal, edge=nidcpower.DigitalEdge.RISING)
```

Use:
```
nidcpower.Session.exported_measure_trigger_output_terminal
```
instead of:
```
export_signal
```

One piece of functionality we removed is configuring the trigger edge (rising vs falling). We did this after careful consideration because all our SMUs are PXI / PXIe, which per the PXI spec defines triggers going over the trigger bus as rising edge. If this is a problem for your application, we'd love to hear more about it.

I'll close the issue but feel free to re-open or reply if not everything was addressed.

closed 2018-07-10 by marcoskirsch

user0129, 2018-07-11:
Hi,

Thanks for the response. It makes sense to me.

Just a last question; we used to have the following "niDCPower_ConfigureSoftwareEdgeMeasureTrigger". Is that not supported now?

Thanks.

marcoskirsch, 2018-07-11:
Refer to [`nidcpower.Session.measure_trigger_type `](http://nimi-python.readthedocs.io/en/master/nidcpower/class.html#measure-trigger-type)