ni/nimi-python #914: error -1074126836 The specified script or sequence is invalid

Pierre-Denis, 2018-08-01:
* Description of issue

    \<Script Generation with FGEN: error -1074126836 The specified script or sequence is invalid.\>

* `python -c "import nifgen; nifgen.print_diagnostic_information()"` output 

    ```
    Name:      NI-FGEN
    Version:   17.1.0.49152
    Module:
        Name:      nifgen
        Version:   1.0.0
    Python:
        Version:   3.6.5 (v3.6.5:f59c0932b4, Mar 28 2018, 17:00:18) [MSC v.1900 64 bit (AMD64)]
        Bits:      64
        Is_Venv:   False
        Installed Packages:
            six==1.11.0
            setuptools==39.0.1
            pip==18.0
            numpy==1.15.0
            niscope==1.0.0
            nifgen==1.0.0
            nidmm==1.0.0
            {'os': {'name': 'Windows', 'version': '6.1.7601', 'bits': '64'}, 'python': {'ver sion': '3.6.5 (v3.6.5:f59c0932b4, Mar 28 2018, 17:00:18) [MSC v.1900 64 bit (AMD64)]', 'bits': '64', 'is_venv': False, 'packages': [{'name': 'six', 'version': '1.11.0'}, {'name': 'setuptools', 'version': '39.0.1'}, {'name': 'pip', 'version': '18.0'}, {'name': 'numpy', 'version': '1.15.0'}, {'name': 'niscope', 'version': '1.0.0'}, {'name': 'nifgen', 'version': '1.0.0'}, {'name': 'nidmm', 'version':'1.0.0'}]}, 'driver': {'name': 'NI-FGEN', 'version': '17.1.0.49152'}, 'module':{'name': 'nifgen', 'version': '1.0.0'}}

    ```
* Portion of the code used having problem:
    ``` python 
        session.output_mode = nifgen.OutputMode.SCRIPT       		
        session.script_triggers[0].script_trigger_type = nifgen.ScriptTriggerType.SOFTWARE_EDGE 
       session.script_triggers[0].digital_edge_script_trigger_edge=nifgen.ScriptTriggerDigitalEdgeEdge.RISING            
        Sine = generate_Sineware() #Function in the file: generate list of floats
        session.channels[0].write_waveform ('sine',Sine)
        session.channels[0].write_script(Script)
        session.script_to_generate = Script
     ```
* Steps to reproduce issue
    ```
    1- Download the ZIP File with Python Code Inside.
    2. Modify the HW ressource currently named as AWG with eg PXISlot7 (name of the ressource)
    3. Launch the code, error should appear. Same script is done with LV and is working.
    ```
* Error code returned:
    ```
     >>py "AWG Script.py"
    Traceback (most recent call last):
      File "AWG Script.py", line 55, in <module> 
        main()
      File "AWG Script.py", line 52, in main
        _main(sys.argv[1:])
      File "AWG Script.py", line 48, in _main
        example(args.resource_name, args.option_string)
      File "AWG Script.py", line 39, in example
        with session.initiate():
      File "C:\Program Files\Python36\lib\site-packages\nifgen\session.py", line 3157, in initiate
        return _Generation(self)
      File "C:\Program Files\Python36\lib\site-packages\nifgen\session.py", line 53,in __init__
        self._session._initiate_generation()
      File "C:\Program Files\Python36\lib\site-packages\nifgen\session.py", line 67,in aux
        return f(*xs, **kws)
      File "C:\Program Files\Python36\lib\site-packages\nifgen\session.py", line 3981, in _initiate_generation
        errors.handle_error(self, error_code, ignore_warnings=False, is_error_handling=False)
      File "C:\Program Files\Python36\lib\site-packages\nifgen\errors.py", line 95,in handle_error
        raise DriverError(code, description)
    nifgen.errors.DriverError: -1074126836: The specified script or sequence is invalid.
    ```      
[AWG Script.zip](https://github.com/ni/nimi-python/files/2248529/AWG.Script.zip)

Is there something wrong with my program or is it wrong behaviour linked with the driver?

Thanks in advance for any help!!!

Bye

P. Denis

marcoskirsch, 2018-08-02:
What NI-FGEN device are you using?

texasaggie97, 2018-08-02:
The problem is on line 36 of the script where you set
``` python
session.script_to_generate = Script
```

This should be
``` python
session.script_to_generate = 'Script'
```

Note the quotes around the name. Without the quotes, you are saying generate a script named `script Script\nrepeat until scriptTrigger0\nGenerate sine\nend repeat\nend script` which doesn't exist.

I'd like to see about taking your script though and turning it into an additional example.

closed 2018-08-02 by texasaggie97