ni/nimi-python #920: Printing waveformInfo type fails with AttributeError

charterchap, 2018-08-20:
* Printing waveformInfo type fails with AttributeError

* 
 capture a waveform on the scope and print(waveform)

self = <[AttributeError("'WaveformInfo' object has no attribute 'actual_samples'") raised in repr()] WaveformInfo object at 0x2592e92c4e0>

    def __repr__(self):
        return '{0}(absolute_initial_x={1}, relative_initial_x={2}, x_increment={3}, actual_samples={4}, offset={5}, gain={6})'.format(
            self.__class__.__name__, self.absolute_initial_x, self.relative_initial_x, self.x_increment,
>           self.actual_samples, self.offset, self.gain)
E       AttributeError: 'WaveformInfo' object has no attribute 'actual_samples'

C:\Program Files\Python36\lib\site-packages\niscope\waveform_info.py:72: AttributeError

Fladolcetta, 2018-08-21:
@charterchap 
Have you tried printing the waveform after you index it? Similar to the following:

https://github.com/ni/nimi-python/blob/1f193aa260ead422188b8f7a81929bd9c12cce60/src/niscope/examples/niscope_fetch.py#L19

When I attempted: `print(waveforms)` I got the same error that you did.
Changing to `print(waveforms[0])` caused me to read the waveform infomation.

```
channel             : 0record              : 0Absolute X0         : 0
Relative X0         : -9.99999e-06
dt                  : 2e-08
offset              : 0
gain                : 1wfm length          : 1,000
```

charterchap, 2018-08-21:
thanks, I see the same with the [0]

Fladolcetta, 2018-08-21:
> thanks, I see the same with the [0]

Do you mean that you see the same as I do and I should close this issue? 

Or did you mean that you see the same error?

charterchap, 2018-08-24:
I would close it. I see the same thing. It would be nice if it printed something other than an error. like <list .... WaveformInfo> but i think having this issue will be enough that people can at least search the issue.

PR #933 (2018-10-11): niscope.WaveformInfo __repr__ and __str__ improvements
- [X] This contribution adheres to [CONTRIBUTING.md](https://github.com/ni/nimi-python/blob/master/CONTRIBUTING.md).
- [X] I've updated [CHANGELOG.md](https://github.com/ni/nimi-python/blob/master/CHANGELOG.md) if applicable.
- [X] I've added tests applicable for this pull request

### What does this Pull Request accomplish?
* Fix formatting when printing a `niscope.WaveformInfo`
* Fix `__repr__` for `niscope.WaveformInfo`

### List issues fixed by this Pull Request below, if any.
* Fix #920

### What testing has been done?
* Unit
* System
closed 2018-10-11 by texasaggie97