ni/nimi-python #926: When NI Switch Executive is added, need to add switch executive string buffer mechanism

Fladolcetta, 2018-08-23:
* Description of issue

In Switch Executive, for strings, you:
1. create a pointer 
2. Assign a value to that pointer with the size of the string buffer you have pre-allocated
3. Pass the pointer into the dll 
4. Read pointer containing the actual length of the string

This is similar to the IVI dance, except

1. The length of the string is not returned from the function. Switch Executive returns the error code always. If I used ivi-dance as a mechanism, then the string size is always 0, resulting in an empty string.
2. The direction is out instead of in

To resolve this for Switch Executive, I had to do the following.

- Change type from pointer to VIInt32 array of size 1 (that is essentially a pointer.)
- Set direction from out, to in (so both I and the dll could set the value, as not setting it would cause no string)
- Set default value of "Size" to the default used in examples,
- Use "Size" to allocated the size of the string buffer.

This enables the string to be read, and partially hidden from the customer (via defaults).

We should create a mechanism for string buffers that use this modified IVI-Dance mechanism.

MrDuff8, 2019-07-19:
This scheme was added to the code generator in the course of work on NI-Digital. It's called "IVI Dance with a Twist".

texasaggie97, 2019-07-24:
> This scheme was added to the code generator in the course of work on NI-Digital. It's called "IVI Dance with a Twist".

This is not actually like `ivi-dance-with-a-twist` like we though because it uses the same parameter as both the in size as well as the out size. This means the `ivi-dance-with-a-twist` code generation we have does not work. We would need another mechanism.

marcoskirsch, 2022-04-14:
NI Switch Executive as a product is in sustaining. If we were to onboard another driver with the same scheme then we'd want to fix it, I think, as the workaround is exceedingly clever (in a bad way).

closed 2022-04-14 by marcoskirsch