ni/nimi-python #927: TypeError when Importing nidcpower on Python 3.7

lkolbly, 2018-09-04:
* Description of issue

When importing nidcpower on Python 3.7 on Windows 10 64-bit, the following stack trace is raised:
```
>>> import nidcpower
Traceback (most recent call last):
  File "<pyshell#0>", line 1, in <module>
    import nidcpower
  File "C:\Users\nitest\AppData\Local\Programs\Python\Python37\lib\site-packages\nidcpower\__init__.py", line 7, in <module>
    from nidcpower.session import Session  # noqa: F401
  File "C:\Users\nitest\AppData\Local\Programs\Python\Python37\lib\site-packages\nidcpower\session.py", line 7, in <module>
    import nidcpower._attributes as _attributes
  File "C:\Users\nitest\AppData\Local\Programs\Python\Python37\lib\site-packages\nidcpower\_attributes.py", line 3, in <module>
    import nidcpower._converters as _converters
  File "C:\Users\nitest\AppData\Local\Programs\Python\Python37\lib\site-packages\nidcpower\_converters.py", line 54, in <module>
    @_convert_repeated_capabilities.register(six.string_types)  # noqa: F811
  File "C:\Users\nitest\AppData\Local\Programs\Python\Python37\lib\functools.py", line 801, in register
    f"Invalid first argument to `register()`: {cls!r}. "
TypeError: Invalid first argument to `register()`: (<class 'str'>,). Use either `@register(some_class)` or plain `@register` on an annotated function.
```

* `python -c "import niscope; niscope.print_diagnostic_information()"` output (replace `niscope` with appropriate package name)

```
C:\Users\nitest\AppData\Local\Programs\Python\Python37>python -c "import nidcpower; nidcpower.print_diagnostic_information()"
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "C:\Users\nitest\AppData\Local\Programs\Python\Python37\lib\site-packages\nidcpower\__init__.py", line 7, in <module>
    from nidcpower.session import Session  # noqa: F401
  File "C:\Users\nitest\AppData\Local\Programs\Python\Python37\lib\site-packages\nidcpower\session.py", line 7, in <module>
    import nidcpower._attributes as _attributes
  File "C:\Users\nitest\AppData\Local\Programs\Python\Python37\lib\site-packages\nidcpower\_attributes.py", line 3, in <module>
    import nidcpower._converters as _converters
  File "C:\Users\nitest\AppData\Local\Programs\Python\Python37\lib\site-packages\nidcpower\_converters.py", line 54, in <module>
    @_convert_repeated_capabilities.register(six.string_types)  # noqa: F811
  File "C:\Users\nitest\AppData\Local\Programs\Python\Python37\lib\functools.py", line 801, in register
    f"Invalid first argument to `register()`: {cls!r}. "
TypeError: Invalid first argument to `register()`: (<class 'str'>,). Use either `@register(some_class)` or plain `@register` on an annotated function.
```
:D

But, in seriousness, it's nidcpower version 1.0.0 ostensibly. Whatever's on pip.

* Steps to reproduce issue

``` python
import nidcpower
```

lkolbly, 2018-09-04:
I suspect it's related to this bug, but haven't looked into it too much: https://bugs.python.org/issue34498

Fladolcetta, 2018-09-04:
Duplicate of #895

closed 2018-09-04 by Fladolcetta