ni/nimi-python #929: Support for niDCPower_InitializeWithChannels

kunaljdoshi, 2018-09-12:
I would like to open multiple sessions for a device (for example a separate session per channel of smu hardware). However, currently only Session class is exposed which does not allow multiple instances. 

  File "c:\source\py3_6_venv\lib\site-packages\nidcpower\errors.py", line 95, in handle_error
    raise DriverError(code, description)
nidcpower.errors.DriverError: -1074135008: IVI: (Hex 0xBFFA0020) Unknown channel or repeated capability name.
teardown: closing session

It would be very helpful to have a way to have mutiple sessions opened; similar to http://zone.ni.com/reference/en-XX/help/370736U-01/ni_dc_power_supplies_help/channelbased/

marcoskirsch, 2018-09-12:
You can most definitely open channel-based Sessions in nidcpower Python module.
See: https://nimi-python.readthedocs.io/en/1.0.0/nidcpower/class.html#nidcpower.Session

I will close. If I misunderstood the question and didn't solve feel free to reopen.

kunaljdoshi, 2018-09-13:
my bad.. I had accidentally simulated a instrument with 1 channel. Thanks for the quick support.

closed 2018-09-13 by kunaljdoshi