ni/nimi-python #934: Classes with dynamic members are user-hostile and maintenance-hostile

epage, 2018-10-11:
Example: WaveformInfo from https://github.com/ni/nimi-python/pull/933

User Problem
- Users can't rely on the code or documentation in front of them to know what they can access
- IDE's can't auto-complete
- mypy will complain

Maintenance problems
- The intent of the code is less clear, making it more likely someone will break it when changing it.

texasaggie97, 2018-10-15:
Documentation is already correct for these members - See return info in https://nimi-python.readthedocs.io/en/master/niscope/class.html#niscope.Session.fetch

marcoskirsch, 2018-10-16:
> Classes with dynamic members are user-hostile and maintenance-hostile #934

Are there examples other than `WaveformInfo`?
IIRC the only reason this class has dynamic members is because of Python 2.7 limitations. Outside of dropping Python 2.7, what would you suggest?

epage, 2018-10-16:
How is this coupled to Python 2.7?

#935 implements exactly what I'd expect.

texasaggie97, 2018-10-16:
`WaveformInfo` ended up being the only "custom type" needed.

The dynamic-ness comes from not Python 2.7, but because we decided to add `channel`, `record` and `samples` to each `WaveformInfo` instead of requiring the user to parse out the portion of the 1D array coming back from the driver that applies to this channel and record.

What you are thinking about regarding Python 2.7 is how `samples` is added. In Python 3+ it is a `memoryview` of the 1D array from the driver to avoid copies. In Python 2.7, `memoryview` doesn't support arrays so we have to create an `array.array` for each `WaveformInfo` by copying the samples from the 1D array in to a new one.

PR #935 (2018-10-16): Cleanup waveform info - add explicit fields used
- [X] This contribution adheres to [CONTRIBUTING.md](https://github.com/ni/nimi-python/blob/master/CONTRIBUTING.md).
- [X] ~~I've updated [CHANGELOG.md](https://github.com/ni/nimi-python/blob/master/CHANGELOG.md) if applicable.~~
- [X] ~~I've added tests applicable for this pull request~~
### What does this Pull Request accomplish?
* Be more explicit with the fields we use in `niscope.WaveformInfo`

### List issues fixed by this Pull Request below, if any.
* Fix #934 

### What testing has been done?
* System
* Unit
* Hand
closed 2018-10-16 by texasaggie97