ni/nimi-python #959: example code in nifgen suggests Linux OS.... tell me more?

bjmuld, 2019-01-15:
The shebang in [nifgen_arb_waveform](https://github.com/ni/nimi-python/blob/master/src/nifgen/examples/nifgen_arb_waveform.py) and probably others have 
`#!/usr/bin/python`
at the head, suggesting a Linux OS.   Every time I ask, NI tech support tells me that NI-FGEN and NI-SCOPE are Windows only drivers.... for a long time now.

Can you guys elucidate me?  Are you in fact running modern NI-SCOPE and NI-FGEN on Linux machines?

epage, 2019-01-15:
Around Christmas time, we released NI-SCOPE for Linux.  See http://www.ni.com/product-documentation/52786/en/

I'm not the person who authored that code but my guess is we were just being consistent on our practices between drivers.

marcoskirsch, 2019-01-16:
> I'm not the person who authored that code but my guess is we were just being consistent on our practices between drivers.

This was the intent. 

Right now there is Linux support for NI-SCOPE, NI-DMM, NI-SWITCH. I can see how it is misleading to follow that pattern in the example for the rest of the drivers. Thanks for reporting.

closed 2019-07-19 by MrDuff8