ni/nimi-python #961: Error changing from OuputMode FUNC to ARB and reverse

ihouses, 2019-01-29:
I have a FGEN with two output channels.

I first set a DC value and then I set a ARB waveform. Somehow after the ARB waveform has finished and abort and close called, the driver gets block and next time that I try to generate the DC value complains in start_generation():

nifgen.errors.DriverError: -1074126845: The operation cannot be completed in the current output mode.

Output mode should be the same for all channels in the session.

If I omit the DC phase and go directly for the ARB generation it works. 
I try to fix the error with the information of the error (Output mode should be the same for all channels in the session.) but I get other errors.

First I set a DC value, setting channel 0 to OuptutMode.FUNC using:

self.session.channels[0].output_mode = nifgen.OutputMode.FUNC
 self.session.channels[0].configure_standard_waveform(wave_dict[type_], amplitude, int(freq),
                                                                        dc_offset, start_phase)
 self.start_generation()

I need to specify the channel, instead of using:
 self.session.output_mode = nifgen.OutputMode.FUNC
 if not I get the error:
nifgen.errors.DriverError: -1074135024: IVI: (Hex 0xBFFA0010) Invalid value for parameter or property.

Property: DC Offset
Channel Name: 0
Invalid Value: 671.0e-3
Minimum Value: 0.0
Maximum Value: 0.0

Doing this:

 self.session.channels[0].output_mode = nifgen.OutputMode.FUNC
 self.session.channels[1].output_mode = nifgen.OutputMode.FUNC

Generates the same error.

marcoskirsch, 2019-01-29:
> I have a FGEN with two output channels.

Please post the exact model you are using.
Please post a full script to reproduce the problem.

Thanks

ihouses, 2019-01-30:
Hello, I paste a script that probably will fail to you too:

Model: 'Model': '5433 (2CH)

The first part of the script works (The ARB mode) but when I change to DC I get the error '
Output mode should be the same for all channels in the session.'
And I can not find a work around for this.

----------------------------------------------------------------------------
import nifgen
from nifgen import enums

options = {'simulate': False, 'driver_setup': {'Model': '5433 (2CH)', 'BoardType': 'PXIe', }, }

session = nifgen.Session(resource_name='FGEN', options=options)

session.abort()

# generate a WAVEFORM
sample_rate = 10E6

waveform = []
for i in range(0, 1000):
    waveform.append(i/1000)

session.output_mode = nifgen.OutputMode.ARB
session.channels[0].arb_sample_rate = sample_rate
session.channels[0].trigger_mode = enums.TriggerMode.SINGLE
waveform_handle = session.create_waveform(waveform_data_array=waveform)

session.exported_start_trigger_output_terminal = "PXI_Trig0"

session.arb_waveform_handle = waveform_handle
session.streaming_waveform_handle = waveform_handle
session.channels[0].configure_arb_waveform(waveform_handle=waveform_handle, gain=2, offset=0)

session.initiate()

while session.is_done() is False:
    pass

session.abort()

#set a DC value
session.channels[0].output_mode = nifgen.OutputMode.FUNC
session.channels[0].configure_standard_waveform(nifgen.Waveform.DC, 1.1, 0, 0, 0)

session.initiate()

# Depending of the internal FGEN state, the script crashes always here like:
#    raise DriverError(code, description)
#    nifgen.errors.DriverError: -1074126845: The operation cannot be completed in the current output mode.

MrDuff8, 2019-01-30:
Hello, 

This issue sounds like it's more related to the NI-FGEN driver (or your specific device) than to the Python API for that driver. For those types of questions, we generally direct you to contact ni.com/support to get the best response and expertise. For this specific issue, I'd encourage you to post to this discussion forum: http://forums.ni.com/t5/Signal-Generators/bd-p/310. You should be able to reference the details in this post, and the reproducing script should help the Applications Engineers find a solution for you. Thanks very much!

closed 2019-01-30 by MrDuff8

ihouses, 2019-01-31:
Hello,
Actually the problem is somehow library related.

Look like that both channels must have the same output mode properties so:
session.channels[0].output_mode = nifgen.OutputMode.FUNC

Should not be available for this propertie.

Instead of choosing the channel the right way would be:

session.output_mode = nifgen.OutputMode.FUNC

Doing so there is no error. Actually I don't know if the second channel is fine or not since I am not using it for now :?