ni/nimi-python #962: Export signal has been removed, options?

ihouses, 2019-01-30:
Some time ago I was using the export signal attribute:

 self.session.export_signal(enums.Signal.START_TRIGGER, "", trigger_name)

Now I see that doesn't exist anymore and checking in the history I see that was done purposely.

I thought that this code would have the same effect:

self.session.exported_start_trigger_output_terminal = trigger_name

But this is not working, if I wait for the trigger that FGEN is generation in the scope never comes.

What is the workaround for the missing export_signal?

marcoskirsch, 2019-01-30:
What you're doing seems correct, it's equivalent. I have verified the NI PXIe-54x3 and can confirm that the implementation of the underlying C function `niFgenMunich_ExportSignal` simply turns around and sets C attribute `NIFGEN_ATTR_EXPORTED_START_TRIGGER_OUTPUT_TERMINAL`.

There is a chance something else is wrong in your program.

* Can you confirm that in the old `nifgen` when you were using `export_signal` triggering worked for you?
* What terminal are you exporting the StartTrigger to?
* What scope are you using? If it's an NI branded scope, how are you configuring it?
* Are you initiating the scope before the signal generator?
* What chassis do you have? What slots are the devices in?
* In general, it is recommended that you _pull_ the signal generator's event from the receiving session rather than export it from the emitting session and let NI's routing subsystem handle trigger line selection for you.

This should give you a good starting point. It would be a lot more helpful if you posted a small script that reproduces the problem. Attach the file or use Markdown syntax for readability. But so far, this does not seem to be a nimi-python bug, so the appropriate place for help are NI's forums.

MrDuff8, 2019-07-19:
We assume this is resolved because we haven't heard back. If this is still an issue, please reopen and attach the requested info.

closed 2019-07-19 by MrDuff8