ni/nimi-python #966: `nidcpower.Session.get_channel_name()` has incorrect documentation, maybe behavior

marcoskirsch, 2019-03-07:
# Description of issue

While perusing the [documentation](https://nimi-python.readthedocs.io/en/master/nidcpower/class.html#get-channel-name) I noticed two odd things:

1. The tip for repeated capabilities is there. This is not a channel based function!
2. The return value (string) is not specified. [Function does appear to return the string](https://github.com/ni/nimi-python/blob/master/generated/nidcpower/session.py).

marcoskirsch, 2019-03-07:
This is a pretty small bug, but I worry that the generated function may  be wrong, and/or that the same codegen bug (if it turns to be a codegen bug as opposed to a metadata bug) shows up in other places.

sbethur, 2021-06-18:
This is not an issue anymore, can be closed.

alexdubois-ni, 2021-06-25:
I can verify that the behavior is correct and thus we can close this issue.

I installed NI-DCPower 20.7 and built the latest nimi_python and locally installed the package. I opened an independent channels session and ran get_channel_name() and it operates how I expect which is the exact same as the C function niDCPower_GetChannelName. You pass in an integer index (which is unfortunately 1-based instead of 0-based...but that's a whole different problem stemming from C) and get back a string with the channel-name. In addition, in a multi-instrument session, this returned channel-name is fully qualified as you'd expect too.

closed 2021-06-25 by marcoskirsch