ni/nimi-python #979: NI MI cannot be called by Python

seuwc1989, 2019-05-13:
* Description of issue
This is Chen Wang from NI China AES team. I am working with a customer to use Python to call NI's MI including DCPower, FGEN, DMM and scope. Now we have a problem that we cannot initialize them. 
 
[image]
[image]
[image]
[image]

marcoskirsch, 2019-05-13:
The `nidcpower` Python module is failing to load the driver runtime. Specifically, it tries to load either `nidcpower_32.dll` or `nidcpower_64.dll` depending on process bitness. That is most likely due to an installation problem. Is this a "clean" system, or is this a system in which a lot of installations / upgrades / downgrades have occurred?

* What OS version / bitness are you using?
* What is the bitness of the Python installation?
* Can the user upgrade NI-DCPower
* Is this problem occurring only with NI-DCPower or with other Modular Instruments drivers too?
* Can the customer run the NI-DCPower Soft Front Panel? This is a 32-bit application.
* Can the customer run InstrumentStudio? This is a 64-bit application.

If you can answer these questions we will get a much better idea about the problem. So far I'm not led to believe that this is a bug in `nimi-python`.

seuwc1989, 2019-05-13:
> The `nidcpower` Python module is failing to load the driver runtime. Specifically, it tries to load either `nidcpower_32.dll` or `nidcpower_64.dll` depending on process bitness. That is most likely [...]

Hi

Thanks for your reply.

I agree with you that this may related with installation as we did a lot of installations / upgrades / downgrades.
> * What OS version / bitness are you using?
Win 7 64bit

> * What is the bitness of the Python installation?
64 bit

> * Can the user upgrade NI-DCPower
We upgrated to 17.5 today but it cannot help.

> * Is this problem occurring only with NI-DCPower or with other Modular Instruments drivers too?
All the MI cannot work. We also upgraded DMM to 17.5 version. It cannot wok as well.

> * Can the customer run the NI-DCPower Soft Front Panel? This is a 32-bit application.
Yes. The SFP can work.

> * Can the customer run InstrumentStudio? This is a 64-bit application.
No. We didn't find InstrumentStudio.

The customer also mentioned that the code can work before. But after some driver upgrade then it cannot work any more. Even we uninstalled and reinstalled to the previous version of driver.

Do you have any suggestion on how to resolve it?

Thank you

marcoskirsch, 2019-05-13:
> Can the customer run the NI-DCPower Soft Front Panel? This is a 32-bit application.
Yes. The SFP can work.

This tells us that the 32-bit driver runtime for the driver is ok. Customer could try running his/her script using 32-bit Python. That will help us narrow things down.

> What is the bitness of the Python installation?
64 bit

This tells us that you are trying to use the 64-bit runtime for the driver.

> Can the customer run InstrumentStudio? This is a 64-bit application.
No. We didn't find InstrumentStudio.

This would help us diagnose whether the problem is with the 64-bit runtime. It's an [optional install](http://www.ni.com/en-us/support/downloads/software-products/download.instrumentstudio.html). The customer should try it, it's very nice.

The 64-bit runtimes are installed in `C:\Program Files\IVI Foundation\IVI\Bin` by default. See if the `nidcpower_64.dll` file is there. If it is there, try [Dependency Walker](http://dependencywalker.com) which will help us see if the problem is a missing dependency.

Note: So far none of this indicates a bug in nimi-python.

marcoskirsch, 2019-05-15:
Closing since there is no indication that this is a bug in nimi-python itself.

closed 2019-05-15 by marcoskirsch