ni/nimi-python #981: niSync support

edupo, 2019-05-14:
Hi!

I was checking for some python binding to the niSync. Then I found this project. Is not clear to me if is already supported or how easy is to integrate... From LabView it seems to me a pretty simple library with some methods that are handled in the same way as other supported libraries.

I'm willing to investigate a bit, but clues on where to start are wellcome.

Thanks in advance!

marcoskirsch, 2019-05-14:
Hello,

nimi-python doesn't include support for the NI-Sync API. The team that owns NI-Sync is distinct from the team that owns Modular Instruments (NI-DMM, NI-SCOPE, etc) so, to be honest, we did not evaluate it and we had not received any requests for it.

Based on the C API ([nisync.h.txt](https://github.com/ni/nimi-python/files/3178230/nisync.h.txt)) it is clear to me that support for it would fit well within the nimi-python APIs because it follows the IVI style of API design.

As far as how to go about this...

The first thing is to provide API description metadata. For the drivers published here, NI has provided this metadata which is created using internal tools. I'd have to do a bit of research in order to find out if everything needed by these internal tools is available for NI-Sync.

The rest of the work consists of makefile / codegen glue, writing some Python-specific API metadata, creating examples, writing system tests, etc.

Given the things on our roadmap, it wouldn't be realistic to expect NI to publish a polished `nisync` python module anytime soon. But if you're willing to roll up your sleeves and do some work, I can investigate providing the API metadata sooner rather than later. It'd be up to you, at least for a while, to do the rest of the work off a development branch at least to get to the point where the module statisfies your immediate needs.

How does this sound?

edupo, 2019-05-15:
Hi Marcos,

That sounds great indeed. I've added `nisync` myself to the `src` folder and copied and modified what seems logic from other modules. From what I read from other metadata files is that `functions.py` is generated from it's `_addon.py` counterpart, so I just removed it from the new module. And then I hit the wall since apparently those are, as you mentioned, generated in some other way...

So I would be glad to further work on this if I can get the metadata of the module generated or I can generate it myself (I could not find anything in the makefiles...).

Thank you again. I'm looking forward to collaborate a bit on this.

MrDuff8, 2019-05-16:
Hello Edu, 

Out of curiosity, can you tell us a little about your application? What are you building and what hardware is in the system? Is using python a requirement or just your preference? Thanks in advance for any additional color.

edupo, 2019-05-17:
Hi Duff,

Silicon validation. We use the PXIe-6674T as clock reference for our clock tests. All our software is developed in Python.

As our functionality requirements are just a couple (creating a route and configuring the DDS frequency) I can wrap the niSync.dll/.so by myself, but is always better to contribute to already existing projects if possible.

We extensively use other packages from this project so I will be glad to help ;)

marcoskirsch, 2019-05-17:
@edupo what version of Python are you currently using?

edupo, 2019-05-27:
Python 2.7

marcoskirsch, 2019-06-05:
Hi @edupo,

> I can investigate providing the API metadata sooner rather than later.

I looked into this and unfortunately there are more complications than I hoped for and the NI-Sync cannot help with it in the short term. This means you will have to wrap niSync library and live with that for the foreseable future.

As far as the nimi-python project is concerned, it's not clear when in the roadmap we'd work on NI-Sync support. But I will leave this issue open and I encourage anyone out there interested in an official `nisync` Python module to chime in so that this is fed back into our prioritization process.

closed 2019-08-21 by MrDuff8