ni/nimi-python #984: Add NI-RFSA support

samcarey, 2019-05-20:
Howdy,

Are there any plans to add NI-RFSA support in the future?
I was starting to write some wrappers in Cython when I found this project which seems to meet a lot of my needs. That's the only API that I need that isn't on the implemented list already.
Excellent work so far! Thanks!

texasaggie97, 2019-05-21:
I have passed on your request but because it is NIWeek this week, any response may be slow. I will update this issue once I get a response.

texasaggie97, 2019-05-22:
The official answer from Brian Glass (Senior Product Manager, RF Test):
“Native NI-RFSA/RFSG API support for Python is not currently available. However, the C# .NET API can be called in a straightforward manner from Python using the Python for .NET library.”

closed 2019-05-22 by texasaggie97

mgunyho, 2019-09-10:
Hi,

I would like to request support for NI-RFSG as well. I'm writing my own crude DLL wrappers, but it would be nice to have official support.

rriola, 2019-09-16:
Hi,

Thank you so much for creating these APIs! I agree with samcarey, are there maybe any future plans on developing Python APIs for Python for RFSA/RFSG?

I'm still trying out the examples from https://github.com/NISystemsEngineering/Python-dotNet-5820/blob/master/source/PythondotNetTest.py ,  and I'll give feedback as well once avaialable.

texasaggie97, 2019-09-16:
I have passed on that there are additional requests but at this point there are no plans for adding NI-RFSA or NI-RFSG support in nimi-python.

mgunyho, 2020-04-09:
In case someone is interested, I've made our wrappers for RFSG available [here](https://github.com/QCoDeS/Qcodes_contrib_drivers/tree/master/qcodes_contrib_drivers/drivers/NationalInstruments). This particular wrapper is based on the [Qcodes](https://github.com/QCoDeS/Qcodes) measurement framework, but the `dll_wrapper` module is standalone. The actual wrapping is done by the `ni_dll_instrument` and `RFSG` modules, which use Qcodes classes, but they should be pretty easy to adapt.