ni/nimi-python #1024: Should we have a context manager for `nitclk.initiate()`?

marcoskirsch, 2019-09-04:
# Description of issue

All the nimi-python APIs we have today (except for `nise` and `nimodinst` where it does not apply) return a context manager from `Session.initiate()` that calls `Session.abort()` on `__exit()__`.

This allows pythonic use of the API. See the code below:

```python
    with niscope.Session(resource_name=resource_name, options=options) as session:
        session.configure_vertical(range=voltage, coupling=niscope.VerticalCoupling.AC)
        session.configure_horizontal_timing(min_sample_rate=50000000, min_num_pts=length, ref_position=50.0, num_records=1, enforce_realtime=True)
        with session.initiate():
            waveforms = session.channels[channels].fetch(num_samples=length)
        for i in range(len(waveforms)):
            print('Waveform {0} information:'.format(i))
            print(str(waveforms[i]) + '\n\n')
```

NI-TClk API has the function

```c
ViStatus _VI_FUNC niTClk_Initiate(ViUInt32 sessionCount, const ViSession sessions[]);
```
... which we intend to present in the `nitclk` Python module as
```python
nitclk.initiate(list_of_session_references)
```

... but there is not corresponding `niTClk_Abort()` function. In Python, this would mean that unless we put in some extra effort, `nitclk` users would need to call abort on each of their sessions like they do in C or LabVIEW which would be inconsistent with the rest of the nimi-python modules. 

The code without the context manager would look something like:

```python
sessions = [my_scope_session, my_fgen_session]
nitclk.initiate([ s.tclk for s in sessions ])
# do stuff...
for s in sessions:
   s.abort()
```
Note the advantage in Python versus C or LabVIEW in which, even if client is forced to abort the sessions, duck-typing is used in order to abort in a loop without caring if the sessions are for different drivers.

# Proposal

If we provide a context manager, the code would look something like:
```python
sessions = [my_scope_session, my_fgen_session]
with nitclk.initiate([ s.tclk for s in sessions ])
   # do stuff...
```
I think this is nicer and more consistent with the rest of nimi-python, but it does go beyond what we usually do because this context manager would not even be calling NI-TClk on `__exit()__`. 

Good idea or bad idea?

epage, 2019-09-04:
I feel the conversation is hampered by not having full example code.  Wouldn't it be:
```python
with contextlib.ExitStack() as stack:
    my_scope_session = stack.enter_context(niscope.Session(resource_name=resource_name, options=options))
    my_fgen_session = stack.enter_context(...)

    sessions = [my_scope_session, my_fgen_session]
    nitclk.initiate([ s.tclk for s in sessions ])

  # do stuff....
```
If so, then why would the user need to call `abort` on each session?  And why would tclk be trying to simplify that for the user?

Also, in your example solution
```python
sessions = [my_scope_session, my_fgen_session]
with nitclk.initiate([ s.tclk for s in sessions ])
   # do stuff...
```
We are passing tclk handles to `initiate` and expecting that to create a context manager that would call `abort` on `__exit__`?  Can it safely call abort with just a tclk handle?

thejcannon, 2019-09-04:
What about just using `contextlib.ExitStack` to manage the list of sessions? I'm not sure I understand how the addition of nitclk muddies the water for session management.

```py
from contextlib import ExitStack

with ExitStack() as stack:
    # or however you want to open sessions in a loop
    sessions = [stack.enter_context(open_session(*params)) for params in session_params]
    nitclk.initiate([ s.tclk for s in sessions ])
    # On we go!
```

homechicken80, 2019-09-05:
For some background nitclk.initiate() is an option convenience function.  Users could just as easily use the initiate on the individual drivers.  The only advantage of the tclk version is it will initiate the master of the start trigger last.

Other pieces of information (not an opinion about your question).
- FYI:  Scope and RFSA doesn't require the user to call abort (handled internally to the driver).  Are you talking about Fgen and RFSG?
-   For mixed instrument use cases (ex. scope and arb) it is common to start all the instruments together using tclk initiate but then only initiate in a loop with the scopes.  The use case is start everything together but then I need to take bursty captures of the data (common in RF stimulus response use cases).
-  Keep in mind that TClk's API is a set of functions that act on sessions.  There is no state, so your abstraction shouldn't imply there is one.  We have discussed the idea of making TClk state based so that 'unsynchronizing or unconfiguring' could be supported.  However, that may require some sort of TClk object/session.
-  I've run into user code that synchronizes devices and then takes the sessions and gives them to separate threads (one per channel) to fetch data.  Would anything you are proposing preclude that use case?

thejcannon, 2019-09-05:
Ah I seemed to have confused `abort` with `close`. `open`/`close` are opposites, as are `initiate`/`abort`.

texasaggie97, 2019-09-05:
@epage The context manager you are invoking is `initialize`/`close` and not `initiate`/`abort`. 

This is a simplified example of existing context managers
``` python
import niscope
with niscope.Session('dev1') as session:  # starts initialize/close
    with session.initiate():  # starts initiate/abort
        session.fetch()
```

As it stands right now (and this is what this question is about), calling initiate via nitclk, does _not_ invoke abort on each of the drivers. So
``` python
with contextlib.ExitStack() as stack:
    my_scope_session = stack.enter_context(niscope.Session(resource_name=resource_name, options=options))
    my_fgen_session = stack.enter_context(...)

    sessions = [my_scope_session, my_fgen_session]
    nitclk.initiate(sessions)
    # do stuff....
```
will not call `my_scope_session.abort()` nor `my_fgen_session.abort()`, but this would:
``` python
with contextlib.ExitStack() as stack:
    my_scope_session = stack.enter_context(niscope.Session(resource_name=resource_name, options=options))
    my_fgen_session = stack.enter_context(...)

    sessions = [my_scope_session, my_fgen_session]
    with nitclk.initiate(sessions):
        # do stuff....
```

marcoskirsch, 2019-09-05:
@epage said:
> I feel the conversation is hampered by not having full example code.

I think you are referring to managing the lifetime of the sessions. In example code it is common to open the session, do something, then closing the session which would call abort for you.
In lots of real-world scenarios it is more common to open the session once and reuse it throughout the tests. This means you will [ configure, initiate, abort, configure, initiate... ] many times without closing the session.

> We are passing tclk handles to initiate and expecting that to create a context manager that would call abort on __exit__? Can it safely call abort with just a tclk handle?

I think so, yes.

marcoskirsch, 2019-09-05:
@homechicken80 said:

> For some background nitclk.initiate() is an option convenience function. Users could just as easily use the initiate on the individual drivers. The only advantage of the tclk version is it will initiate the master of the start trigger last.

I expect customers to use `nitclk.initiate(sessions)` 99.99% of the time rather than "just as easily" figure out who the master is, call initiate on the slaves, then on the master. Similarly, the user does not _have to_ use the context manager we return from `Session.initialize()` in the individual APIs, but I expect them to most of the time. 

> FYI: Scope and RFSA doesn't require the user to call abort (handled internally to the driver). Are you talking about Fgen and RFSG?

This is only try for non-continuous operations that are eventually done, right? But even though it is not required to call `abort()`, it is nice to have a context manager that will do so for you, for example, if you exit the scope due to an exception being raised.

> For mixed instrument use cases (ex. scope and arb) it is common to start all the instruments together using tclk initiate but then only initiate in a loop with the scopes. The use case is start [...]

This is great info, I think we should strongly consider providing an example of this.

> I've run into user code that synchronizes devices and then takes the sessions and gives them to separate threads (one per channel) to fetch data. Would anything you are proposing preclude that use case?

I don't think this proposed API does anything to preclude that.

thejcannon, 2019-09-05:
If you can address the principal of least surprise, then I think the proposal is fine.

As it stands though, the user is going to assume the `__exit__` will do the opposite method __on the object__ it was `__enter__`'ed with. So the fact that `with nitclk.initiate()` doesn't call `nitclk.abort()` when cleaning up is surprising.

Changing the API to bring some attention to the discrepancy might work (give it a different name). Or adding `nitclk.abort` to the Python API (and documenting it doesn't exist on the underlying API).

texasaggie97, 2019-09-05:
Something I thought of that may make this moot.

Based on one of the decisions in #1005, the `sessions` parameter is an iterable where each element is one of:
1. An integer that represents the NI-TClk session for that driver session
2. An `nitclk.SessionReference`
3. An object with a `tclk` property which is an `nitclk.SessionReference`

Only 3 has the possibility of working, and that is if it has an `abort()` method. Neither of the other two have enough information to even figure out _how_ to call `abort()`

marcoskirsch, 2019-09-05:
>Something I thought of that may make this moot.
>
>Based on one of the decisions in #1005, the sessions parameter is an iterable where each element is one of:
>
>An integer that represents the NI-TClk session for that driver session
An nitclk.SessionReference
An object with a tclk property which is an nitclk.SessionReference
Only 3 has the possibility of working, and that is if it has an abort() method. Neither of the other two have enough information to even figure out how to call abort()

I think we can reconsider that decision. From Zen of Python:

>There should be one-- and preferably only one --obvious way to do it.

For clients, that would be number 3 (not I modified my original code snippet for number 3).
Number 1 is for non-nimi-python drivers that want to use nimi-python's `nitclk` module. Those don't exist and we can address it when we need to.
I'm not sure number 2 provides advantages over number 3, but maybe you can tell me why it does.

So let's assume

* We do number 3
* There is a Python API for an NI product that supports NI-TClk and wants to work with `nitclk` module.

The requirements would be:
* Have an object with a property called `tclk`
* **if** you want to use the `niclk.initiate()` context manager, you must have an `abort()` function.

If the API cannot provide `abort()` then simply don't use the context manager.

texasaggie97, 2019-09-05:
Accepting any of the three options also applies to attributes. So these all work:
``` python
scope1 = niscope.Session('dev1')
scope2 = niscope.Session('dev2')

scope1.tclk.scrip_trigger_master_session = scope2._vi
scope1.tclk.scrip_trigger_master_session = scope2..tclk
scope1.tclk.scrip_trigger_master_session = scope2
```
Likewise, getting the attribute will return a `niscope.SessionReference`:
``` python
scope1 = niscope.Session('dev1')
scope2 = niscope.Session('dev2')

ms = scope1.tclk.script_trigger_master_session
assert type(ms) == nitclk.SessionReference

ms.sample_clock_delay = 42
```

marcoskirsch, 2020-01-22:
Discussed in code review.

* NI-TClk doesn't have an abort/stop function
* Session passed in may call the abort/stop function something else, we'd need to standardize on somehting.
* `nitclk` functions may take a SessionReference which doesn't have start/stop. We'd need to solve that.

We decided not to do it, which means that client will need to call `abort` on all the sessions.

closed 2020-01-22 by marcoskirsch