ni/nimi-python #1062: Wrong implementation of get / set attribute calls

jerome-st, 2019-10-30:
* Description of issue

When use of some embedded variable as serial_number, it returns errors

* `python -c "import nidigital; nidigital.print_diagnostic_information()"` 

nidigital version 0.1.0

* Steps to reproduce issue

python -c "import nidigital; session=nidigital.Session(resource_name='PXI1Slot4');print(session.serial_number)"
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "C:\WPy-3680\Win64Python36SMD\python-3.6.8.amd64\lib\site-packages\nidigital\_attributes.py", line 81, in __get__
    return session._get_attribute_vi_string(self._attribute_id)
AttributeError: 'Session' object has no attribute '_get_attribute_vi_string' 

* Fix proposal

at line 81 of _attributes.py file, use get_attribute_vi_string instead of _get_attribute_vi_string
same issue is seen for all get and set attribute of the different types.

sbethur, 2019-10-30:
Thanks for bringing this issue to our notice. We are working on the fix. In the meantime, you can use this workaround:

`session.get_attribute_vi_string(1150001)`

[nidigital's session](https://github.com/ni/nimi-python/blob/master/generated/nidigital/session.py#L505) can be used to figure out the attributeId.

Similarly, if you want to get/set channel-based attributes, you can do:
```
# AttributeId 1150013 is active_load_ioh
session.channels[0].set_attribute_vi_real64(1150013, -0.0015)
active_load_ioh = session.get_attribute_vi_real64(1150013)
```

closed 2019-11-06 by texasaggie97

sbethur, 2019-11-06:
@jerome-st , The fix has been pushed to master. You can get the fix by building nidigital.

We are planning to make another pre-1.0 release for nidigital in 2 weeks that will include this fix and some other fixes and changes. Please let us know if you can wait till then. Otherwise we can make another release with just this fix now.

PR #1087 (2019-11-06): Fix get/set properties in nidigital
- [x] This contribution adheres to [CONTRIBUTING.md](https://github.com/ni/nimi-python/blob/master/CONTRIBUTING.md).

- [x] I've updated [CHANGELOG.md](https://github.com/ni/nimi-python/blob/master/CHANGELOG.md) if applicable.

- [x] I've added tests applicable for this pull request

### What does this Pull Request accomplish?

Fix get/set properties in nidigital by making get/set attribute methods in Session private.

### List issues fixed by this Pull Request below, if any.

* Fix #1062

### What testing has been done?

Added system tests for get/set prope [... cut]