ni/nimi-python #1064: Enum for NIDIGITAL_ATTR_HISTORY_RAM_MAX_SAMPLES_TO_ACQUIRE_PER_SITE is missing from nidigital API

sbethur, 2019-10-30:
* Description of issue

In the Digital C header, there is:
```
/*- Values for NIDIGITAL_ATTR_HISTORY_RAM_MAX_SAMPLES_TO_ACQUIRE_PER_SITE --*/
#define NIDIGITAL_VAL_ACQUIRE_ALL_SAMPLES       -1
```

Enum for this is missing from nidigital enums.py

texasaggie97, 2019-11-01:
All enums (`NIDIGITAL_VAL_*`) in the header are in the metadata.

If an enum is not used anywhere (parameter type or attribute type) then it will not be in the generated code. The fix for this issue is to set `'enum'` in the metadata appropriately.

sbethur, 2019-11-01:
Ah, gotcha. We have an [issue](https://github.com/ni/nimi-python/issues/1066) open for updating methods and properties to use enums. Looks like there's nothing to be done for this issue, I'll close it.

closed 2019-11-01 by sbethur