ni/nimi-python #1065: Remove redundancy in nidigital API

sbethur, 2019-10-30:
* Description of issue

According to Zen of Python:

> There should be one-- and preferably only one --obvious way to do it.

In nidigital API, for few things, there is more than one way of accomplishing it.
e.g.:

1. Pin control mode can be selected via `selected_function` property or `select_function` method.
2. History RAM configuration
3. PPMU configuration
4. Levels configuration

Note that the above list is not exhaustive.

API should be reviewed and redundancy removed, when possible.

marcoskirsch, 2019-10-30:
For configuration that is accessible via attribute or function in the C API, on Python we have exposed only the property.

We leave the function only in cases in which a single function groups several closely related attributes.

sbethur, 2019-11-07:
> We leave the function only in cases in which a single function groups **_several_** closely related attributes. [emphasis added]

What about functions that group two closely related attributes?
e.g.:
```
niDigital_PPMU_ConfigureCurrentLimit(behavior, limit)
niDigital_PPMU_ConfigureVoltageLimits(lowerVoltageLimit, upperVoltageLimit)
```

PR #1189 (2020-01-15): Remove redundant methods from the nidigital API
- [x] This contribution adheres to [CONTRIBUTING.md](https://github.com/ni/nimi-python/blob/master/CONTRIBUTING.md).

- [x] I've updated [CHANGELOG.md](https://github.com/ni/nimi-python/blob/master/CHANGELOG.md) if applicable.

~- [ ] I've added tests applicable for this pull request~

### What does this Pull Request accomplish?

Methods removed:
```
ConfigureCycleNumberHistoryRAMTrigger
ConfigureDigitalEdgeConditionalJumpTrigger
ConfigureDigitalEdgeStartTrigger
ConfigureFirstFailureHistoryRAMTrigger
ConfigureHistoryRAMCyclesToAcquire
ConfigurePa [... cut]
closed 2020-01-15 by marcoskirsch