ni/nimi-python #1067: Multi-session methods are missing from nidigital API

sbethur, 2019-10-30:
These multi-session methods are missing from the API:

- niDigital_BurstPatternSynchronized
- niDigital_WriteSequencerFlagSynchronized
- niDigital_SortPinResultsBySiteViReal64
- niDigital_SortPinResultsBySiteViInt64
- niDigital_SortPinResultsBySiteViUInt8
- niDigital_SortSiteResultsViBoolean
- niDigital_SortSiteResultsViUInt32Waveform

These methods need to accept array of session handles.

marcoskirsch, 2019-10-31:
If these methods are superseded by multi-instrument sessions then we should not add them to the brand new python API.

sbethur, 2019-11-11:
> If these methods are superseded by multi-instrument sessions then we should not add them to the brand new python API.

There is one use-case that would not be covered by multi-instrument sessions - If the user wants to tclk multi-instrument nidigital session with another driver session. This use-case was brought up during development of multi-instrument session support, with other driver most-likely being an RF driver. But any solution (other than just using single-instrument session) was tabled with the understanding that it's not a common use-case. If there's a real need for it, the solution would ideally be within multi-instrument session support.

Given that the above use-case is not common and it could be solved lower in the stack, I think we should not add these methods to Python API 1.0. We can always add them later.

@marcoskirsch , @texasaggie97 Do you have any concerns?

marcoskirsch, 2019-11-11:
> If the user wants to tclk multi-instrument nidigital session with another driver session.

This does not work. NI-Digital multi-instrument sessions cannot be used along with NI-TClk. If/when the driver runtime supports it then we can revisit.

sbethur, 2019-11-11:
> This does not work. NI-Digital multi-instrument sessions cannot be used along with NI-TClk. 

Sorry, I meant if users need to tclk multiple digital sessions and some other driver session, they cannot use multi-instrument session, and hence would need to use the above methods in question.

texasaggie97, 2019-11-11:
Would they have to use the multi-instrument sessions? Or can they just use nitclk like they would with any other mix of sessions?

sbethur, 2019-11-11:
> Would they have to use the multi-instrument sessions? Or can they just use nitclk like they would with any other mix of sessions?

They can use nitclk to synchronize mixed sessions. But then they would need to use:
- `niDigital_BurstPatternSynchronized` - If they want to synchronize the burst across session. Although they can achieve synchronization using `niTClk Initiate` and some other digital methods to do what `niDigital_BurstPatternSynchronized` does. So, this is really a convenience method.
- `niDigital_WriteSequencerFlagSynchronized` - If they want to synchronize some control flow within patterns.
- `niDigital_SortPinResultsBySiteViReal64`, `niDigital_SortPinResultsBySiteViInt64`,  `niDigital_SortPinResultsBySiteViUInt8`, `niDigital_SortSiteResultsViBoolean`, `niDigital_SortSiteResultsViUInt32Waveform` - If they want a convenient way to combine results from multiple sessions.

mlopezleon, 2020-01-08:
I think we should wait until a use-case comes up. I'm ok with not adding these multi-session methods to the python API.

closed 2020-01-08 by sbethur