ni/nimi-python #1068: niDigital_EnableMatchFailCombination is missing from nidigital API

sbethur, 2019-10-30:
This is a multi-session method like [these](https://github.com/ni/nimi-python/issues/1067), but it also accepts a NI-Sync session handle.

marcoskirsch, 2022-04-14:
NI-Digital now supports multi-instrument sessions which handles the NI-Sync driver internally. This is the recommended usage. So we won't add `niDigital_EnableMatchFailCombination` to `nidigital` anymore.

ni-jfitzger, 2023-09-18:
>NI-Digital now supports multi-instrument sessions which handles the NI-Sync driver internally. This is the recommended usage. So we won't add niDigital_EnableMatchFailCombination to nidigital anymore.

It's actually NI-TClk that's handled internally by the multi-instrument session. Whether using a multi-instrument session or multiple individual sessions NI-Sync will be called by the NI-Digital API, but opening and closing the NI-Sync session is still handled by the NI-Sync API.

Whether using a multi-instrument session or multiple individual sessions with NI-TClk, niDigital_EnableMatchFailCombination is still the entrypoint that a user would call to do whatever it does. There is no alternative entrypoint, so I'm repening this.

We should maybe only support calling it from a multi-instrument session, though.

ni-jfitzger, 2023-09-18:
Note that this function still wouldn't be useful without a way to open an NI-Sync session, so it will remain unplanned, for now.

doczhivago, 2023-10-05:
@ni-jfitzger Here is a potential workaround for the NI-Sync session:

https://github.com/ni/miv-examples/tree/main/src/NI-SyncPython

closed 2025-02-19 by ni-jfitzger