ni/nimi-python #1069: Single-session methods missing in nidigital API

sbethur, 2019-10-30:
These two methods are missing:
- niDigital_WriteSourceWaveformSiteUniqueU32 (One of the arguments is an IN jagged array)
- niDigital_FetchCaptureWaveformU32 (One of the arguments is an OUT jagged array)

marcoskirsch, 2019-10-31:
@sbethur to split this up into two issues.

texasaggie97, 2019-11-01:
@sbethur, Is this issue only to enable this method or to make a "fancy" version of it?

If fancy, what should the parameter list look like, and what should the output be and in what format?

sbethur, 2019-11-01:
> If fancy, what should the parameter list look like, and what should the output be and in what format?

Updated issue description.

sbethur, 2019-11-13:
Rethinking the types for this method.. we should consider `site_list` and `waveform_data` into a dict or namedtuple. It should consistent with what's used in `fetch_capture_waveform`.

PR #1119 (2019-11-19): Add fancy write_source_waveform_unique
- [X] This contribution adheres to [CONTRIBUTING.md](https://github.com/ni/nimi-python/blob/master/CONTRIBUTING.md).
- [X] I've updated [CHANGELOG.md](https://github.com/ni/nimi-python/blob/master/CHANGELOG.md) if applicable.
- [X] I've added tests applicable for this pull request
### What does this Pull Request accomplish?
* Add fancy wrapper for `write_source_waveform_unique()`

### List issues fixed by this Pull Request below, if any.
* Fix #1069 

### What testing has been done?
* Visual inspection
closed 2019-11-19 by texasaggie97