ni/nimi-python #1073: Need nidigital examples

sbethur, 2019-10-30:
* Description of issue

Need nidigital examples

rriola, 2020-01-08:
Hi nimi-python Team,

Are there any updates on this? 
We'll be looking forward to your most favorable response, and Happy New Year!

sbethur, 2020-01-08:
We are currently finalizing [nidigital API](https://github.com/ni/nimi-python/milestone/20). After that's done, we'll be adding examples, which is tied to [nidigital 1.0](https://github.com/ni/nimi-python/milestone/22) milestone.

As we're working on the API, our goal is to make it as user-friendly and "Pythonic" as possible. As a result, we are envisioning adding examples only for specific parts of the API, where they are warranted.

For now, you can refer to the [system tests](https://github.com/ni/nimi-python/tree/master/src/nidigital/system_tests) we've written and continue to write. Something like the [fetch_capture_waveform test](https://github.com/ni/nimi-python/blob/3e5b0e12c0d6070c36a4c142e7544cb84095f365/src/nidigital/system_tests/test_system_nidigital.py#L182) performs most of the core digital operations (load pattern, load pin map and other files; configure source and capture waveforms; and burst a pattern). It would be awesome if you can start using the API and provide feedback on API usability and need for examples.

PR #1545 (2021-02-03): Create nidigital_ppmu_source_and_measure example
- [x ] This contribution adheres to [CONTRIBUTING.md](https://github.com/ni/nimi-python/blob/master/CONTRIBUTING.md).

~- [ ] I've updated [CHANGELOG.md](https://github.com/ni/nimi-python/blob/master/CHANGELOG.md) if applicable.~

~- [ ] I've added tests applicable for this pull request.~

### What does this Pull Request accomplish?

Add nidigital_ppmu_source_and_measure.py example and its accompanying digital pattern editor generated files to show how to configure and source/measure voltage/current using the PPMU on selected channels/pins of th [... cut]
PR #1547 (2021-02-09): Create nidigital_configure_time_set_and_voltage_levels example
- [x] This contribution adheres to [CONTRIBUTING.md](https://github.com/ni/nimi-python/blob/master/CONTRIBUTING.md).

~- [ ] I've updated [CHANGELOG.md](https://github.com/ni/nimi-python/blob/master/CHANGELOG.md) if applicable.~

~- [ ] I've added tests applicable for this pull request.~

### What does this Pull Request accomplish?

Add nidigital_configure_time_set_and_voltage_levels .py example and its accompanying digital pattern editor generated files to show how to configure time set and voltage levels + burst a pattern on the  [... cut]
PR #1548 (2021-02-10): Delete nidigital_do_nothing example
- [x] This contribution adheres to [CONTRIBUTING.md](https://github.com/ni/nimi-python/blob/master/CONTRIBUTING.md).

~- [ ] I've updated [CHANGELOG.md](https://github.com/ni/nimi-python/blob/master/CHANGELOG.md) if applicable.~

~- [ ] I've added tests applicable for this pull request.~

### What does this Pull Request accomplish?

Delete nidigital_do_nothing.py example since we are adding real nidigital examples 

### List issues fixed by this Pull Request below, if any.

* Fix #1073 

### What testing has been done?
closed 2021-02-03 by marcoskirsch