ni/nimi-python #1074: nidigital fetch_capture_waveform_u32 returns empty list

alexp-st, 2019-10-31:
* Description of issue

In nidigital driver, calling fetch_capture_waveform_u32 method after executing a pattern with digital capture returns an empty list.

I managed to fix this issue by modifying session.py line 3058 as follow:
<- data_buffer_size_ctype = _visatype.ViInt32(error_code)  # case S180
-> data_buffer_size_ctype = _visatype.ViInt32(actual_num_waveforms_ctype.value * actual_samples_per_waveform_ctype.value)

* `python -c "import niscope; niscope.print_diagnostic_information()"` output (replace `niscope` with appropriate package name)

nidigital version: 0.1.0

* Steps to reproduce issue

    1. Step 1: Initialise nidigital session
    2. Step 2: load pinmap, spec, levels, timings, pattern, digital capture
    3. Step3: Execute pattern with digital capture (session.burst_pattern)
    4. Step4: Get digital capture (fetch_capture_waveform_u32)

alexp-st, 2019-11-07:
If my understanding of the driver if correct, this issue should be considered as a BUG.
data_buffer_size_ctype is not correctly updated before the 2nd call to niDigital_FetchCaptureWaveformU32

sbethur, 2019-11-07:
@alexp-st , Yes, it is a bug. There is [PR](https://github.com/ni/nimi-python/pull/1096) in review to fix this API entry point.

PR #1112 (2019-11-18): Add fancy fetch capture waveform
- [X] This contribution adheres to [CONTRIBUTING.md](https://github.com/ni/nimi-python/blob/master/CONTRIBUTING.md).
- [X] I've updated [CHANGELOG.md](https://github.com/ni/nimi-python/blob/master/CHANGELOG.md) if applicable.
- [X] ~~I've added tests applicable for this pull request~~

### What does this Pull Request accomplish?
* Replaces #1096
* Create a "fancy" wrapper around `fetch_capture_waveform_u32()`
    * Returns array of named tuples Measurement(data, site)
    * Gets site list and uses that when creating named tuple

### List issues fixed by this Pu [... cut]
closed 2019-11-18 by marcoskirsch