ni/nimi-python #1079: In nidigital Session.__init__(), get rid of id_query argument

sbethur, 2019-10-31:
# Description of issue

In C API, documentation for `IDQuery` argument to `niDigital_InitWithOptions` says:

> A Boolean that verifies that the digital pattern instrument you initialize  is supported by NI-Digital. NI-Digital automatically performs this query, so  setting this parameter is not necessary.

Seems like we can remove this argument from Session.__init__().

marcoskirsch, 2019-11-13:
We should, none of the other nimi-python drivers expose this because it has no effect on NI drivers. It is an IVI consideration applicable to boxed instruments.

texasaggie97, 2019-11-14:
We don't actually remove them because that would mess up the code generation to call into the DLL. What we do is give these types of parameters a default value. This is already done for nidigital.
``` python
def __init__(self, resource_name, id_query=False, reset_device=False, options={}):
   ...
        self._vi = self._init_with_options(resource_name, id_query, reset_device, options)
```

sbethur, 2019-11-14:
Can we make the method fancy and truly remove it from the parameter list?

marcoskirsch, 2019-11-14:
Or change the code-generator so that we can specify parameters to remove from the public API and default values for the underlying ctypes call into the Library?

texasaggie97, 2020-01-07:
Do we really want to remove them?

If we remove them for all drivers we may break customers that have set them.
If we only remove them for nidigital, then nidigital is different from the rest.

sbethur, 2020-03-19:
@marcoskirsch, should we move this to "2.0 Source Breakers" milestone?

marcoskirsch, 2020-03-23:
Shreyas, yes let’s do that.