ni/nimi-python #1088: Tooling required to allow users to easily find support files for nidigital examples

sbethur, 2019-11-05:
### Description of issue

nidigital examples require support files (pattern files, pinmap file etc.). Examples are not installed on the system when a nimi-python API is installed, rather users find the examples on [ReadTheDocs](https://nimi-python.readthedocs.io/en/master/nidcpower/examples.html).

Example page on ReadTheDocs should point the users to the support files location and explain where they should copy the support files, before running the example.

texasaggie97, 2019-12-18:
I created two solutions for this because I couldn't decide which I liked better.
#1162 - I think this is a better solution for the user, but the solution is complicated and messy with obscure make content
#1163 - Maybe less convenient for the user but the solution is clean and compact

sbethur, 2020-01-06:
My vote is for per-driver zip files. Usually, users writing test programs will target one or two APIs at a time.

PR #1163 (2020-01-08): Create per driver zip files for examples
- [X] This contribution adheres to [CONTRIBUTING.md](https://github.com/ni/nimi-python/blob/master/CONTRIBUTING.md).
- [X] I've updated [CHANGELOG.md](https://github.com/ni/nimi-python/blob/master/CHANGELOG.md) if applicable.
- [X] ~~I've added tests applicable for this pull request~~

### What does this Pull Request accomplish?
* Add a zip file per driver for examples and any helper files
* Add link to zip file in examples documentation

### List issues fixed by this Pull Request below, if any.
* Fix #1088 

### What testing has been done?
* Visual ins [... cut]
closed 2020-01-08 by marcoskirsch