ni/nimi-python #1092: Bounds checking not done for int values

sbethur, 2019-11-05:
### Description of issue

Bounds checking is not done for int type method arguments and property values. It appears that an int value larger than the underlying ctypes type gets silently truncated and passed down to the C driver DLL. It would be nice if bounds checking is done in the Python layer.

### System report

`python -c "import niscope; niscope.print_diagnostic_information()"` output

```
OS:
    Name:      Windows
    Version:   10.0.16299
    Bits:      64
Driver:
    Name:      NI-SCOPE
    Version:   19.0.0.49152
Module:
    Name:      niscope
    Version:   1.1.3
Python:
    Version:   3.6.8 (tags/v3.6.8:3c6b436a57, Dec 24 2018, 00:16:47) [MSC v.1916 64 bit (AMD64)]
    Bits:      64
    Is_Venv:   True
    Installed Packages:
        zipp==0.6.0
        wcwidth==0.1.7
        six==1.12.0
        pytest==5.2.1
        pyparsing==2.4.2
        py==1.8.0
        pluggy==0.13.0
        packaging==19.2
        numpy==1.17.3
        nitclk==0.1.0.dev0
        niscope==1.1.3
        nidigital==0.1.1.dev0
        nidcpower==1.1.2
        more-itertools==7.2.0
        importlib-metadata==0.23
        future==0.18.1
        enum34==1.1.6
        colorama==0.4.1
        attrs==19.3.0
        atomicwrites==1.3.0
        setuptools==40.8.0
        pip==19.0.3
```

### Steps to reproduce issue

```
with niscope.Session(resource_name='my_device', options='Simulate=1, DriverSetup=Model:5164; BoardType:PXIe') as session:
    # Type of tv_trigger_line_number is ViInt32. val is 1 followed by 32 0s.
    val = 4294967296
    session.tv_trigger_line_number = val
    assert session.tv_trigger_line_number == 0
```

Run the above snippet and observe the incorrect value in `assert`

texasaggie97, 2019-11-06:
Should this error checking be done in all attributes and in all functions every time? I think: if we want to ever do the check, we need to always do the check

Are we concerned about any effect on performance these extra checks will add? I think: In general, if you care about performance, maybe Python isn't for you...

What error/exception should we return? I think: OverflowError

sbethur, 2019-11-06:
I agree that it should be always or never. We could provide a way to turn off bounds checking for users who want to squeeze out some performance.

Also, we should benchmark and figure out what's the performance hit before deciding one way or the other.

Overflow seems to be the right exception to throw.