ni/nimi-python #1094: Add Python 3.8 to our test suites

marcoskirsch, 2019-11-06:
#Description of issue

Python 3.8 was released last month but our CI does not have coverage for it. What if something that worked just fine doesn't work on 3.8? How will we catch it? See #1091 

We need to add this to both Travis CI unit tests and nimi-bot system tests.

marcoskirsch, 2019-11-06:
https://github.com/ni/nimi-python/commit/087f452e700ec107369b77afb030304719416ccf

Makes it so unit tests use released Python 3.8 and breaks on failure.

texasaggie97, 2019-11-06:
numpy doesn't yet support Python 3.8. See https://github.com/scipy/scipy/issues/10927

marcoskirsch, 2019-11-06:
> numpy doesn't yet support Python 3.8. See scipy/scipy#10927

How does this affect us? We don't try to use numpy unless the customer is using it. Is the lack of numpy breaking our automated tests?

texasaggie97, 2019-11-07:
> How does this affect us? 

It breaks system tests on Python 3.8.

For customers, as long as they don't use numpy they will be fine since we don't declare an explicit dependency on it.

marcoskirsch, 2019-11-11:
The question for us then is, do we:

* skip all system tests on Python 3.8 until numpy support is official
* instrument the tests to skip specific numpy-dependent tests and then remove that when numpy support is official.

texasaggie97, 2020-02-07:
Unit tests (Travis CI) and system tests (Jenkins/nimi-bot) now all test on Python 3.8

closed 2020-02-07 by texasaggie97