ni/nimi-python #1124: Remove programmatic pin map creation API from nidigital

marcoskirsch, 2019-11-14:
Programmatic pin map creation is a very niche need that needs a lot of functions and testing. The vast majority of users load pin maps that they create in the PBD / TSM pin map editors or using a text editor.

We should not expose such niche functionality unless there is a very compelling reason to do so.

sbethur, 2020-01-08:
TL;DR: Yes, we can remove programmatic pin map creation API.

Discussed this with the Digital driver team..
- We are not breaking any use-case if we don't provide the programmatic API
- We are not aware of any customer workflow that requires programmatic API
- Programmatic pinmap API was added in Digital 0.9 as a stop-gap before the loading of pinmap file was made available in 1.0
- It's always easier to add than remove

Given these points, we can remove the programmatic pinmap creation API.

For the sake of completeness, we also discussed about other similar "low-level" API for configuring levels and timesets. We want to keep these API, even though there is load-file way, because it's common for levels/timeset values to be changed during a test program, i.e. Load a set of levels/timeset values from file and then modify one of more values during test execution as required.

marcoskirsch, 2020-01-08:
So we can close #1186 ?

sbethur, 2020-01-08:
> So we can close #1186 ?

Yes

closed 2020-01-21 by marcoskirsch

markusroeleke, 2020-03-10:
We at Bosch want to use the programmatic API to generate the PinMap. I now have to write an generator to generate the PinMap in order to load it to the card. We want to reconfigure the PXI6570 dynamically during runtime of the test program. We have a couple of cards here. Please add this functionality again.

sbethur, 2020-03-10:
@markusroeleke, First off, thanks for using the pre-release nidigital API and giving us feedback before we finalize the API. We want to understand your usecase better and have a few questions:

1. Can you please elaborate on how you're dynamically reconfiguring 6570 during runtime?
    - Are you closing and reopening the session to load a new pinmap or are you calling `reset()`?
    - Does the runtime dynamic reconfiguration also involve changes to the physical test setup like changing the loadboard/DUT, rewiring the connections, etc.?
2. How many pinmaps are being used in the test program and how different are they?
3. What kind of DUTs are being tested?
4. Is the test program for characterization or production test?
5. Are you using TestStand Semiconductor Module?

marcoskirsch, 2020-03-10:
@markusroeleke please open a new issue and add as much context as possible. Thanks!

PR #1213 (2020-01-21): Remove programmatic pin map creation API from nidigital
- [x] This contribution adheres to [CONTRIBUTING.md](https://github.com/ni/nimi-python/blob/master/CONTRIBUTING.md).

- [x] I've updated [CHANGELOG.md](https://github.com/ni/nimi-python/blob/master/CHANGELOG.md) if applicable.

~- [ ] I've added tests applicable for this pull request~

### What does this Pull Request accomplish?

Remove programmatic pin map creation API from nidigital.

### List issues fixed by this Pull Request below, if any.

* Fix #1124 

### What testing has been done?

Visual inspection of generated files