ni/nimi-python #1132: Fancy functions should support data types other than array.array

texasaggie97, 2019-11-18:
### Description of issue
When adding fancy functions (specifically `write_source_waveform_unique()` so far) we need to consider what other types we want to accept for the input.

Right now, it is just `array.array("L")`

marcoskirsch, 2019-11-18:
To add a bit more detail:

This is applicable to `nidigital` and maybe other places. Specifically, the way the code is written as of #1119 being merged, is that the client must provide the digital waveform as an `array.array`. This is very un-pythonic. A Python developer would reasonably have the expectation that any other iterable type of numbers should work.

In fact, that is precisely the case in other `nimi-python` fancy functions such as `[nifgen.Session.write_waveform](https://github.com/ni/nimi-python/blob/master/generated/nifgen/session.py#L2944)`.

Hopefully we can make this work without too much hand-written code by leveraging parameter converters.

sbethur, 2019-11-21:
When fixing this issue, please un-skip the test cases for numpy.array and list in [test_source_waveform_parallel_site_unique](https://github.com/ni/nimi-python/blob/c7f66d9e49b49110cf4c7d3dd769e9398d435e39/src/nidigital/system_tests/test_system_nidigital.py#L123)

PR #1184 (2020-01-17): Allow more types for the waveform in nidigital.write_source_waveform_site_unique()
- [X] This contribution adheres to [CONTRIBUTING.md](https://github.com/ni/nimi-python/blob/master/CONTRIBUTING.md).
- [X] I've updated [CHANGELOG.md](https://github.com/ni/nimi-python/blob/master/CHANGELOG.md) if applicable.
- [X] I've ~~added~~updated tests applicable for this pull request
### What does this Pull Request accomplish?
* Adds support to `write_source_waveform_site_unique()` for numpy.array and list.
### List issues fixed by this Pull Request below, if any.
* Fix #1132 

### What testing has been  [... cut]
closed 2020-01-17 by marcoskirsch