ni/nimi-python #1135: "Null address for New Option String

InvncibiltyCloak, 2019-11-20:
### Description of issue

    <Replace with a description of the issue>

### System report

`python -c "import niscope; niscope.print_diagnostic_information()"` output (replace `niscope` with appropriate package name)

    ```
    <Replace with the output of the above command>
    ```

### Steps to reproduce issue

    ``` python
    <Replace with a script that clearly shows the issue, if applicable>
    ```

    1. Step 1
    2. Step 2
    3. etc...

InvncibiltyCloak, 2019-11-20:
Accidentally submitted this immediately. I am working on updating the bug contents. Sorry for any inconvenience.

marcoskirsch, 2019-11-27:
Can you try a newer runtime version of NI-DCPower?

[NI-DCPower 19.5](https://www.ni.com/en-us/support/downloads/drivers/download.ni-dcpower.html#329028) is the latest released on 10/25/19.

marcoskirsch, 2019-11-27:
Also, what device model are you using?

marcoskirsch, 2020-05-01:
@InvncibiltyCloak any progress on this?