ni/nimi-python #1141: Stop suggesting contributors use MinGW

marcoskirsch, 2019-11-21:
In [CONTRIBUTING.md](https://github.com/ni/nimi-python/blob/master/CONTRIBUTING.md) we include instructions for building on Linux using [MinGW](http://www.mingw.org/Welcome_to_MinGW_org).

Given that Windows 10 is widely adopted and the project can be built using Windows Subsystem for Linux, we should simplify the combination of build options and eliminate the need for this third-party component. I think explaining how to set up the system for building using Windows Subsystem for Linux should be enough.

It would not hurt to review the whole file and improve on it as part of this exercise.

sbethur, 2019-11-21:
I can do this, since I setup the build/test env. recently.

sbethur, 2022-01-31:
More reasons for stopping support for MinGW..

A new contributor to nimi-python ran into multiple issues in trying to setup their dev environment with minGW:
1. A really old version of `hacking` (0.5.4) package gets installed on Windows and leads to this error when building:
```
File "C:\dev\git\github\nimi-python\.tox\64\py39-build_test\lib\site-packages\hacking\core.py", line 185
except ImportError, exc:
^
SyntaxError: invalid syntax
```
2. Running `nidigital` unit tests on Windows results in an error:
```
generated\nidigital\nidigital\unit_tests\test_nidigital.py:102: 
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
generated\nidigital\nidigital\session.py:3777: in __init__
    self.tclk = nitclk.SessionReference(self._vi)
.tox\64\py36-test\lib\site-packages\nitclk\session.py:161: in __init__
    self._library = _library_singleton.get()
.tox\64\py36-test\lib\site-packages\nitclk\_library_singleton.py:50: in get
    ctypes_library = ctypes.CDLL(_get_library_name())
```
Some investigation revealed that running unit tests need NI-TClk driver runtime to be installed on the system, but that's not the case on Windows. This warrants a separate issue.
3. Invoking `find` command on Windows, as part of `codegen` step is returning `FIND: Parameter format not correct`. It appears this has no impact on the build output, but not certain.
4. Certain known warnings during generation of docs on Linux are treated as errors on Windows:
```
c:\dev\git\github\nimi-python\docs\nidigital\class.rst:29: ERROR: Content block expected for the "note" directive; none found.
c:\dev\git\github\nimi-python\docs\nidigital\class.rst:25: WARNING: image file not readable: nidigital/note.gif
c:\dev\git\github\nimi-python\docs\nidigital\class.rst:25: WARNING: image file not readable: nidigital/note.gif
c:\dev\git\github\nimi-python\docs\niscope\class.rst:1872: WARNING: Bullet list ends without a blank line; unexpected unindent.
c:\dev\git\github\nimi-python\docs\niscope\class.rst:218: ERROR: Unknown target name: "meas".
```

closed 2022-09-21 by ni-jfitzger