ni/nimi-python #1153: Lots of `Windows fatal exception: code 0xe06d7363` in our system tests

marcoskirsch, 2019-11-27:
### Description of issue

Not sure when this started happening. I noticed it while looking at console output of nimi-bot tests.

For example, the following test run "passed" but if you look at [nimi_python_system_tests/1561](http://nimipy-jenkins.amer.corp.natinst.com:8080/job/nimi_python_system_tests/1561/console) you will see almost 2000 prints like:

```
Current thread 0x00001740 (most recent call first):
  File "c:\jenkins\workspace\nimi_python_system_tests\.tox\32\py36-system_tests\lib\site-packages\nise\_library.py", line 129 in niSE_OpenSession
  File "c:\jenkins\workspace\nimi_python_system_tests\.tox\32\py36-system_tests\lib\site-packages\nise\session.py", line 762 in _open_session
  File "c:\jenkins\workspace\nimi_python_system_tests\.tox\32\py36-system_tests\lib\site-packages\nise\session.py", line 236 in __init__
  File "C:\Jenkins\workspace\nimi_python_system_tests\src\nise\system_tests\test_system_nise.py", line 7 in session
  File "c:\jenkins\workspace\nimi_python_system_tests\.tox\32\py36-system_tests\lib\site-packages\_pytest\fixtures.py", line 783 in call_fixture_func
  File "c:\jenkins\workspace\nimi_python_system_tests\.tox\32\py36-system_tests\lib\site-packages\_pytest\fixtures.py", line 957 in pytest_fixture_setup
  File "c:\jenkins\workspace\nimi_python_system_tests\.tox\32\py36-system_tests\lib\site-packages\pluggy\callers.py", line 187 in _multicall
  File "c:\jenkins\workspace\nimi_python_system_tests\.tox\32\py36-system_tests\lib\site-packages\pluggy\manager.py", line 87 in <lambda>
  File "c:\jenkins\workspace\nimi_python_system_tests\.tox\32\py36-system_tests\lib\site-packages\pluggy\manager.py", line 93 in _hookexec
  File "c:\jenkins\workspace\nimi_python_system_tests\.tox\32\py36-system_tests\lib\site-packages\pluggy\hooks.py", line 286 in __call__
  File "c:\jenkins\workspace\nimi_python_system_tests\.tox\32\py36-system_tests\lib\site-packages\_pytest\fixtures.py", line 908 in execute
  File "c:\jenkins\workspace\nimi_python_system_tests\.tox\32\py36-system_tests\lib\site-packages\_pytest\fixtures.py", line 579 in _compute_fixture_value
  File "c:\jenkins\workspace\nimi_python_system_tests\.tox\32\py36-system_tests\lib\site-packages\_pytest\fixtures.py", line 498 in _get_active_fixturedef
  File "c:\jenkins\workspace\nimi_python_system_tests\.tox\32\py36-system_tests\lib\site-packages\_pytest\fixtures.py", line 482 in getfixturevalue
  File "c:\jenkins\workspace\nimi_python_system_tests\.tox\32\py36-system_tests\lib\site-packages\_pytest\fixtures.py", line 472 in _fillfixtures
  File "c:\jenkins\workspace\nimi_python_system_tests\.tox\32\py36-system_tests\lib\site-packages\_pytest\fixtures.py", line 293 in fillfixtures
  File "c:\jenkins\workspace\nimi_python_system_tests\.tox\32\py36-system_tests\lib\site-packages\_pytest\python.py", line 1439 in setup
  File "c:\jenkins\workspace\nimi_python_system_tests\.tox\32\py36-system_tests\lib\site-packages\_pytest\runner.py", line 364 in prepare
  File "c:\jenkins\workspace\nimi_python_system_tests\.tox\32\py36-system_tests\lib\site-packages\_pytest\runner.py", line 119 in pytest_runtest_setup
  File "c:\jenkins\workspace\nimi_python_system_tests\.tox\32\py36-system_tests\lib\site-packages\pluggy\callers.py", line 187 in _multicall
  File "c:\jenkins\workspace\nimi_python_system_tests\.tox\32\py36-system_tests\lib\site-packages\pluggy\manager.py", line 87 in <lambda>
  File "c:\jenkins\workspace\nimi_python_system_tests\.tox\32\py36-system_tests\lib\site-packages\pluggy\manager.py", line 93 in _hookexec
  File "c:\jenkins\workspace\nimi_python_system_tests\.tox\32\py36-system_tests\lib\site-packages\pluggy\hooks.py", line 286 in __call__
  File "c:\jenkins\workspace\nimi_python_system_tests\.tox\32\py36-system_tests\lib\site-packages\_pytest\runner.py", line 207 in <lambda>
  File "c:\jenkins\workspace\nimi_python_system_tests\.tox\32\py36-system_tests\lib\site-packages\_pytest\runner.py", line 234 in from_call
  File "c:\jenkins\workspace\nimi_python_system_tests\.tox\32\py36-system_tests\lib\site-packages\_pytest\runner.py", line 207 in call_runtest_hook
  File "c:\jenkins\workspace\nimi_python_system_tests\.tox\32\py36-system_tests\lib\site-packages\_pytest\runner.py", line 182 in call_and_report
  File "c:\jenkins\workspace\nimi_python_system_tests\.tox\32\py36-system_tests\lib\site-packages\_pytest\runner.py", line 90 in runtestprotocol
  File "c:\jenkins\workspace\nimi_python_system_tests\.tox\32\py36-system_tests\lib\site-packages\_pytest\runner.py", line 81 in pytest_runtest_protocol
  File "c:\jenkins\workspace\nimi_python_system_tests\.tox\32\py36-system_tests\lib\site-packages\pluggy\callers.py", line 187 in _multicall
  File "c:\jenkins\workspace\nimi_python_system_tests\.tox\32\py36-system_tests\lib\site-packages\pluggy\manager.py", line 87 in <lambda>
  File "c:\jenkins\workspace\nimi_python_system_tests\.tox\32\py36-system_tests\lib\site-packages\pluggy\manager.py", line 93 in _hookexec
  File "c:\jenkins\workspace\nimi_python_system_tests\.tox\32\py36-system_tests\lib\site-packages\pluggy\hooks.py", line 286 in __call__
  File "c:\jenkins\workspace\nimi_python_system_tests\.tox\32\py36-system_tests\lib\site-packages\_pytest\main.py", line 270 in pytest_runtestloop
  File "c:\jenkins\workspace\nimi_python_system_tests\.tox\32\py36-system_tests\lib\site-packages\pluggy\callers.py", line 187 in _multicall
  File "c:\jenkins\workspace\nimi_python_system_tests\.tox\32\py36-system_tests\lib\site-packages\pluggy\manager.py", line 87 in <lambda>
  File "c:\jenkins\workspace\nimi_python_system_tests\.tox\32\py36-system_tests\lib\site-packages\pluggy\manager.py", line 93 in _hookexec
  File "c:\jenkins\workspace\nimi_python_system_tests\.tox\32\py36-system_tests\lib\site-packages\pluggy\hooks.py", line 286 in __call__
  File "c:\jenkins\workspace\nimi_python_system_tests\.tox\32\py36-system_tests\lib\site-packages\_pytest\main.py", line 246 in _main
  File "c:\jenkins\workspace\nimi_python_system_tests\.tox\32\py36-system_tests\lib\site-packages\_pytest\main.py", line 196 in wrap_session
  File "c:\jenkins\workspace\nimi_python_system_tests\.tox\32\py36-system_tests\lib\site-packages\_pytest\main.py", line 239 in pytest_cmdline_main
  File "c:\jenkins\workspace\nimi_python_system_tests\.tox\32\py36-system_tests\lib\site-packages\pluggy\callers.py", line 187 in _multicall
  File "c:\jenkins\workspace\nimi_python_system_tests\.tox\32\py36-system_tests\lib\site-packages\pluggy\manager.py", line 87 in <lambda>
  File "c:\jenkins\workspace\nimi_python_system_tests\.tox\32\py36-system_tests\lib\site-packages\pluggy\manager.py", line 93 in _hookexec
  File "c:\jenkins\workspace\nimi_python_system_tests\.tox\32\py36-system_tests\lib\site-packages\pluggy\hooks.py", line 286 in __call__
  File "c:\jenkins\workspace\nimi_python_system_tests\.tox\32\py36-system_tests\lib\site-packages\_pytest\config\__init__.py", line 92 in main
  File "c:\jenkins\workspace\nimi_python_system_tests\.tox\32\py36-system_tests\lib\site-packages\py\test.py", line 4 in <module>
  File "c:\jenkins\workspace\nimi_python_system_tests\.tox\32\py36-system_tests\lib\site-packages\coverage\execfile.py", line 192 in run_python_file
  File "c:\jenkins\workspace\nimi_python_system_tests\.tox\32\py36-system_tests\lib\site-packages\coverage\execfile.py", line 122 in run_python_module
  File "c:\jenkins\workspace\nimi_python_system_tests\.tox\32\py36-system_tests\lib\site-packages\coverage\cmdline.py", line 627 in do_run
  File "c:\jenkins\workspace\nimi_python_system_tests\.tox\32\py36-system_tests\lib\site-packages\coverage\cmdline.py", line 491 in command_line
  File "c:\jenkins\workspace\nimi_python_system_tests\.tox\32\py36-system_tests\lib\site-packages\coverage\cmdline.py", line 756 in main
  File "C:\Jenkins\workspace\nimi_python_system_tests\.tox\32\py36-system_tests\Scripts\coverage.EXE\__main__.py", line 7 in <module>
  File "c:\Python36-32\Lib\runpy.py", line 85 in _run_code
  File "c:\Python36-32\Lib\runpy.py", line 193 in _run_module_as_main
Windows fatal exception: code 0xe06d7363
```

Full console output:
[nimi_python_system_tests-1561.txt](https://github.com/ni/nimi-python/files/3898878/nimi_python_system_tests-1561.txt)

This seems to affect all drivers except for NI-TClk and lots of different functions. Trying to get the list
[... 82 lines cut ...]
* `niScope_InitiateAcquisition`
* `niScope_GetError`
* `niScope_AutoSetup`
* `niScope_GetAttributeViInt32`
* `niScope_Disable`
* `niSwitch_InitWithTopology`
* `niSwitch_SetAttributeViInt32`
* `niSwitch_GetError`
* `niModInst_OpenInstalledDevicesSession`
* `niSE_OpenSession`
* `niSE_Connect`

Full command output:
[grep resutls.txt](https://github.com/ni/nimi-python/files/3898883/grep.resutls.txt)

This includes the following modules:

* `nidcpower`
* `nidigital`
* `nidmm`
* `nifgen`
* `nimodinst`
* `niscope`
* `niswitch`
* `nise`

marcoskirsch, 2020-01-08:
Of interest: https://bugs.python.org/issue31701

> Whenever a C++ exception is thrown in an extension the faulthandler dumps a traceback to it, even if it is caught.

Need to read more on this, but it sounds like whenever a C++ exception is thrown inside the DLL, we get this output. This even if the exception is handled. I may try to debug the driver to see where C++ exception gets thrown and evaluate that.

CPython changed the behavior re: faulthandler dumps a traceback (https://bugs.python.org/msg306355). It is from 2017-11-16. It's possible that we have Python installations that are _that_ old. That's another thing for me to check.

marcoskirsch, 2020-01-09:
Confirmed that upgrading Python interpreters makes these traces disappear. @texasaggie97 is working on #1143 which requires setting up new VMs for running system tests. Those will use new versions so we decided to leave the nimi-bot VM we have now alone.

texasaggie97, 2020-02-07:
Moving to new nimi-bot has fixed this since we are using a newer versions of Python

closed 2020-02-07 by texasaggie97