ni/nimi-python #1156: Weird spacing in docstrings

marcoskirsch, 2019-12-05:
### Description of issue

Looking at some of the docstrings, there are weird spacing issues.

[For example](https://github.com/ni/nimi-python/blob/master/generated/niscope/session.py#L141):
```
    Specifies the source the digitizer monitors for an arm reference trigger.   When the arm reference trigger is received, the digitizer begins looking for a  reference (stop) trigger from the user-configured trigger source.
```

There are 3 spaces after the period, 2 spaces between "for a" and "reference".

AlexHearnNI, 2020-03-10:
This looks like a duplicate of Issue #817. For reference, I fixed this for niscope with internal changelists 43576120 and 43576154.

marcoskirsch, 2020-03-11:
Will close. Thanks!

closed 2020-03-11 by marcoskirsch