ni/nimi-python #1157: nimi-bot uses passwords, must use PAT

marcoskirsch, 2019-12-09:
### Description of issue

Received the following email from GitHub to nimi-bot's email:

>Hi @nimi-bot,
>
>You recently used a password to access an endpoint through the GitHub API using Java/1.8.0_66. We will deprecate basic authentication using password to this endpoint soon:
>
>https://api.github.com/repositories/91020334/pulls?state=open
>
>We recommend using a personal access token (PAT) with the appropriate scope to access this endpoint instead. Visit https://github.com/settings/tokens  for more information.
>
>Thanks,
>The GitHub Team

If we don't fix this "soon", then nimi-bot will lose access to the project which will cripple our CI system tests.

marcoskirsch, 2020-01-03:
@texasaggie97 notes:

>From the [GitHub pull request builder plugin](https://wiki.jenkins.io/display/JENKINS/GitHub+pull+request+builder+plugin): You should probably migrate to [GitHub Branch Source Plugin](https://wiki.jenkins.io/display/JENKINS/GitHub+Branch+Source+Plugin)

This indicates that they have no plan on fixing the Jenkins plugin we use on nimi-bot.

texasaggie97, 2020-02-07:
We have switch the new nimi-bot to username/PAT

closed 2020-02-07 by texasaggie97