ni/nimi-python #1178: Forks of nimi-python repo do not get system tests CI

marcoskirsch, 2020-01-03:
### Description of issue

The way nimi-bot operates, it polls `ni/nimi-python` repo for PRs and when it sees one it executes the system tests.

That means that if a contributor forks the repo, they do not get automatic system test execution by nimi-bot; and they need to know to run them locally on a machine with all the drivers installed. The bar for this is very high and will be even higer once #1139 is addressed.

Ideally, our CI system for running system tests would work for forks as well out of the box.

marcoskirsch, 2020-01-23:
@texasaggie97 can at least one of us members of the org trigger tests on PRs from forks?
That way we can look at the diff to make sure it's safe before we execute arbitrary code from an external contributor in nimi-bot which sits inside the company's network.

texasaggie97, 2020-01-23:
Member of the org can not give permission for a test run from a out of org fork. Only admins.

texasaggie97, 2020-02-07:
When a PR is created from a fork back to our repo, the trigger plugin on Jenkins will add a comment asking for permission to run the tests. An admin will need to allow the tests before they run. This is a security precaution so that arbitrary code is not ran.

I misread your question above. If a member of the org creates a fork and then a PR back, then their tests will run, as long as they are a member of the configured org ('ni' right now).

marcoskirsch, 2022-04-14:
PRs from a fork do get it pending admin approval. This is preferable in order to avoid abuse.

closed 2022-04-14 by marcoskirsch