ni/nimi-python #1196: nimi-bot logs aren't world-viewable

marcoskirsch, 2020-01-14:
Our nimi-bot system that runs system tests automatically sits inside NI's network. That means that in a PR you can see the pass/fail result of the tests, but if you click on the link in order to debug what failed, it will be unreachable unless you have access to the company's network.

This is counter to the open source nature of the nimi-python project that makes contributions from non NI employees a lot more challenging.

marcoskirsch, 2020-01-14:
Fortunately, we've designed nimi-bot to use only simulated hardware, so while we need the drivres to be installed, we don't need physical NI hardware to be present.

* One option would be to exposing the Jenkins port to the outside world. There are security implications to this.
* Another option would be to host the machine outside of the network using some cloud provider. There are ongoing costs to doing so and somebody would need to administer patches in a regular basis.
* There might be some other solution in which the test results get uploaded to some world-accessible server. This increases overall complexity but might be doable.

marcoskirsch, 2020-01-14:
In the meantime, contributors can run system tests on a local system that has the NI driver runtimes installed:

    python3 -m tox -e py36-system_tests

Note the precise invokation may vary based on which version of Python you're using, and we have a change queued up that may change the above yet a bit more.

SukruthGrandhiNI, 2021-12-07:
> Fortunately, we've designed nimi-bot to use only simulated hardware, so while we need the drivres to be installed, we don't need physical NI hardware to be present.
>
> * One option would be to [...]

Now moved Jenkins based agents to Azure agents, though we moved to Azure agents security implications remains same. 
Unless we move agents to cloud(Non NI network domain) logs are not accessible for external contributors.

marcoskirsch, 2022-04-14:
The GitHub actions-based nimibot has world-viewable logs.
thanks @SukruthGrandhiNI !

marcoskirsch, 2022-04-14:
@SukruthGrandhiNI we thought using the AW-hosted VMs + GitHub actions for `nimibot` would address this. However, it doesn't seem to work.

Can you take a look?

sukruthG, 2022-04-18:
@marcoskirsch, I could see GitHub action logs when I logged in as external contributor.

[image]

sukruthG, 2022-04-18:
I have made a comment using external contributor credentials.

sukruthG, 2022-04-18:
@marcoskirsch 
@sbethur  is also able to see the logs when he logged in as external contributor.

marcoskirsch, 2022-04-18:
Ok, closing. Thanks for confirming.

closed 2022-04-18 by marcoskirsch