ni/nimi-python #1201: Configure Rising/Falling Edge trigger

MadMax1597, 2020-01-16:
### Description of issue

    Hi, this is regarding this closed issue: https://github.com/ni/nimi-python/issues/902
Here, the reasoning for the removal of configurable Rising/Falling edge trigger functionality is given as:
"One piece of functionality we removed is configuring the trigger edge (rising vs falling). We did this after careful consideration because all our SMUs are PXI / PXIe, which per the PXI spec defines triggers going over the trigger bus as rising edge. If this is a problem for your application, we'd love to hear more about it"
But all the other NI Modular Instruments' Python API have this functionality available. Moreover, there are ways to configure this using the LabVIEW DCPower API. 
Could you elaborate on why exactly this reasoning is given? By all accounts, it seems that the PXI bus can handle both rising and falling edge triggers.

### System report

`python -c "import niscope; niscope.print_diagnostic_information()"` output (replace `niscope` with appropriate package name)

    ```
    <Replace with the output of the above command>
    ```

### Steps to reproduce issue

    ``` python
    <Replace with a script that clearly shows the issue, if applicable>
    ```

    1. Step 1
    2. Step 2
    3. etc...

marcoskirsch, 2020-01-16:
In short: we couldn’t come up with a real world use case for it, so decided to err on the side of keeping the API smaller. Negative logic triggering is more commonly used for interfacing with third party instruments outside of PXI via the front panel connectors.

Do you have a need for configuring the triggers on SMUs as falling edge?

Man-Marius, 2023-02-24:
Hi, 
I need this functionality to be able to trigger on the Failing edge. Is there a work around for this?

Man-Marius, 2023-02-24:
> In short: we couldn’t come up with a real world use case for it, so decided to err on the side of keeping the API smaller. Negative logic triggering is more commonly used for interfacing with third [...]

I really need this functionality. Can you please add it back to the library

marcoskirsch, 2023-02-24:
I would like to hear about your use case and why you need triggers configure in this manner so that the work can be prioritized ahead of other things.

There is an easy workaround.

You can call `_set_attribute_vi_int32(self, attribute_id, attribute_value)`(https://github.com/ni/nimi-python/blob/a94b20c47624b4b91e2b7e3befa313c76543c03d/generated/nidcpower/nidcpower/session.py#L6389) directly to configure any attribute.

I attach `nidcpower.h` header file which has the attribute definitions.

For calculating `attribute_id`, use 1,150,000 for `NIDCPOWER_ATTR_BASE`.
`NIDCPOWER_VAL_BASE` is 1,000.

[nidcpower.h.zip](https://github.com/ni/nimi-python/files/10828809/nidcpower.h.zip)

Man-Marius, 2023-02-27:
> I would like to hear about your use case and why you need triggers configure in this manner so that the work can be prioritized ahead of other things.
> 
> There is an easy workaround.
> 
> You can call `_set_attribute_vi_int32(self, attribute_id, attribute_value)`(
> 
> https://github.com/ni/nimi-python/blob/a94b20c47624b4b91e2b7e3befa313c76543c03d/generated/nidcpower/nidcpower/session.py#L6389
> 
> ) directly to configure any attribute.
> I attach `nidcpower.h` header file which has the attribute definitions.
> 
> For calculating `attribute_id`, use 1,150,000 for `NIDCPOWER_ATTR_BASE`. For `attribute_value` use `NIDCPOWER_VAL_BASE` of 1,000.
> 
> [nidcpower.h.zip](https://github.com/ni/nimi-python/files/10828809/nidcpower.h.zip)

Thanks for the quick reply, but is the NIDCPOWER_ATTR_BASE and NIDCPOWER_VAL_BASE values common for all SMUs devices?

Man-Marius, 2023-02-27:
Hi @marcoskirsch, our user case is to trigger on the falling edge, because that is where the response to the input is taking place. Our DUT measures system changes on the Raising edge while on the Falling edge it applies its response to the system changes.  

By the way, is it possible to configure back panel triggers of the chassis to be used for measuring triggers? I have tried to use the back panel for this purpose but it fails while the front panel trigger on the controller works.

marcoskirsch, 2023-02-27:
> By the way, is it possible to configure back panel triggers of the chassis to be used for measuring triggers? I have tried to use the back panel for this purpose but it fails while the front panel trigger on the controller works.

To make sure I understand, let me paraphrase what you are asking:

What "back panel triggers of the chassis" are you referring to? What is the model of PXIe chassis that you are using? Some NI chassis have a connector in the back for getting triggers across the PXI Trigger bus from one chassis to another one. But I don't know which specific connectors you are referring to.

Man-Marius, 2023-09-07:
> `NIDCPOWER_VAL_BASE`

Hi Marcoskirsch, the advice you gave me last time worked for the PXIe-4141 and now we migrated to the PXIe-4163 SMUs, but now this feature is not working since the migration. I get the error  -1074097882: Requested value is not a supported value for this property.  Can you help me here.

marcoskirsch, 2023-09-07:
While the API is common for NI-DCPower, that does not mean that all instruments support all the same configuration options.
In this case, it appears the NI PXIe-4163 SMU does not support the setting you want. Can you tell me, using the C symbol names from `nidcpower.h` what attribute you are configuring to what value? Maybe a code snippet?

Man-Marius, 2023-09-08:
`
     
        self.__smu._set_attribute_vi_int32(
            attribute_id=(NIDCPOWER_ATTR_BASE + 35),
            attribute_value=(NIDCPOWER_VAL_BASE + 17))
`

bases  are 
`

     NIDCPOWER_ATTR_BASE = 1150000

     NIDCPOWER_VAL_BASE = 1000

 `

ni-jfitzger, 2024-04-25:
After further internal discussion, we determined that this property is not currently supported by any of our newer architecture NI-DCPower devices. Given the complexity and size of the nidcpower API and the fact that the `set_attribute` workaround is readily available, we've decided not to add this property to our public API.

If support for the property is added for a new device, we will revisit that decision.
If the workaround is untenable to a particular user, please comment to let us know.

closed 2024-04-25 by ni-jfitzger