ni/nimi-python #1215: Verify remote debugging from DPE works with nidigital API

sbethur, 2020-01-20:
### Description of issue

Verify remote debugging from DPE works with nidigital API

marcoskirsch, 2020-01-21:
Be aware of #906 asn you try this out. A driver runtime bug in the implementation of remote debugging will force you to turn off "Using Breakpoints in C/C++/.NET Applications".

sbethur, 2020-03-31:
Verified that remote debugging from DPE works as expected.

I looked into adding a test for Remote Debugging in nidigital, but there isn't an easy way to test Remote Debugging in an automated fashion. I didn't dig too deep since there isn't any custom logic in nidigital module specifically for Remote Debugging. The plumbing for it is code generated and is the same for all nimi-python modules. The code generator and generated code are unit-tested in nifake.

closed 2020-03-31 by sbethur