ni/nimi-python #1220: RFC: Change how nimi-python APIs expose triggers and events

marcoskirsch, 2020-01-21:
What follows is a proposal that would change how triggers and events are exposed by the nimi-python modules.

This applies to `nidigital`, `nidcpower`, `nifgen`, `niscope`.
Further analysis is needed for `nidmm` and `niswitch`, whose underlying driver runtime are inconsistent regarding how triggers and events work – or perhaps those inconsistencies first need to be addressed by the underlying driver runtimes.

## How it is now

In the currently shipping modules, per our [API desing guidelines](https://github.com/ni/nimi-python/wiki/API-design-guidelines), we have a "Close 1-to-1 mapping of the C APIs". Each driver has a `Session` class that contains a property for each attribute in the C API, and a method for each function in the C API.

In the C API, triggers are configured through a combination of attributes and functions. The functions have special inputs to specify the trigger being used. But attributes are repeated for each trigger and only use "repeated capability" (aka "channel") input in order to differentiate among instances of the same trigger type.

In the Python APIs repeated capabilities are exposed as "repeated capability containers" that are a property of the `Session` class. This is true for triggers too:

```python
session.script_triggers[1].send_software_edge_trigger() # method
session.script_triggers[1].exported_script_trigger_output_terminal = x # redundant

session.send_software_edge_trigger() # method, defaults to StartTrigger
#session.exported_script_trigger_output_terminal = x # which script trigger? error
session.exported_start_trigger_output_terminal = y # ok
```

* This doesn't scale well if you have a trigger other than "StartTrigger" that is single instance.
* Goes against [Zen of Python](https://www.python.org/dev/peps/pep-0020/) tenet "Explicit is better than implicit" because of StartTrigger being the default sometimes.

There are no functions associated with events. But there are plenty of attributes and for them in the C API we created lots of attributes with different IDs and constant names rather than leverage the repeated capability parameter (for good reason). In Python they are mapped one-to-one from the C counterparts:

```python
session.data_marker_events[2].data_marker_event_data_bit_number = x # redundancy in name
session.done_event_output_terminal = y # unfriendly to code reuse because of unique name
```

## What I'm proposing

This proposal is to group trigger functions and properties under a trigger members of `Session`. These may be a single instance or an indexable instance. These objects would only contain the applicable properties and methods, which would increase usability and make IDE auto-complete a lot more helpful.

```python
session.script_triggers[1].send_software_edge_trigger # same as before.
session.script_triggers[1].output_terminal = x # no more redundancy

session.start_trigger.send_software_edge_trigger() # no more implicit
session.start_trigger.output_terminal = y # no more redundancy
```

Similarly, `Event` object would be made member of `Session` and group the corresponding properties using shortened names.

```python
session.data_marker_events[2].data_bit_number = x # no more redundancy
session.data_marker_events[2].output_terminal = y # no more redundancy
session.done_event.output_terminal = z
```

## What would be needed

* Needs to be timed with a 2.0 source breaker.
* Metadata lists triggers/events API supports.
* Metadata for attributes and functions would need to indicate which triggers/events they act on.
* Documentation would need to be figured out
* New classes Trigger and Event would need to be written. Possibly, multiple classes would need to be generated.
* Code-generator would need to stop placing the event/trigger related functions/attributes in `Session`.

This proposal adds complexity to the Python APIs and moves them farther away from the C/LabVIEW counterparts.
On the other hand, usability is increased and it makes them more similar to the C# counterparts.

sbethur, 2020-01-21:
+1 for the proposal

texasaggie97, 2020-01-23:
Like you say, the proposal is moving further away from C/LabVIEW. One consequence of this is increased maintenance, even if it is code generated. 

What if we removed the "fancy" `send_software_edge_trigger()` completely. It would then be called more like it is in C:
``` python
session.send_software_edge_trigger(nifgen.enums.START_TRIGGER, 'None')
session.send_software_edge_trigger(nifgen.enums.SCRIPT_TRIGGER, 'scriptTrigger0')
```
From a maintenance and code generator this is simpler. It would still be a source breaker though at least for nifgen, would need to wait until 2.0. nidigital though would just stay as it is now without the fancy `send_software_edge_trigger()`.

marcoskirsch, 2020-01-23:
I think this is a good idea, what we have now is a weird in-between. We either do it all the way (my proposal above) or don't (your proposal in the previous message).

sbethur, 2020-02-27:
Another proposal:
- Make nidigital behave like C API i.e. `send_software_edge_trigger` take in the enum for trigger type
- Update nifgen to make it like nidigital, without breaking source compatibility using clever defaults and return a warning when users use it the old way

sbethur, 2020-02-27:
Based on the agreed-upon solution above, I've created two issues:
#1193 
#1300 

@marcoskirsch, can you please close this issue.

closed 2020-03-19 by marcoskirsch