ni/nimi-python #1266: There is more than one tox INI supporting running of system tests and one is incorrect

marcoskirsch, 2020-02-12:
We use `tox` to run system tests.
If a user wants to run system tests, they can either:

(1)
```
python3 -m tox -e niswitch_system_tests
```
 which uses the environments from the top level `tox.ini` file, or

(2)
```
pushd generated\niswitch & python3 -m tox -c tox-system_tests.ini & popd 
```
which uses the code-generated `tox-system_tests.ini` file.

nimi-bot uses (2) when running system tests.
The reason why support wasn't removed from the top level `tox.ini` file is that Automated Test Suites internal to NI rely on the top-level INI file.

Problems with this:
* Top level `tox.ini` is incorrect. For example, it is missing the `fastening` dependency on the modules that need it.
* We aren't exercising those environments in nimi-python CI, which allows for bugs like the above one to go unnoticed.
* Two ways of doing the same thing is confusing to humans.
   * Why two?
   * How are they different?
   * Which one should I be using?

I think we need to get rid of method (1). This will simplify that `tox.ini` and eliminate the redundancy. But this will require internal NI Automated Test Suites to make a change.

marcoskirsch, 2020-02-12:
@injaleea can you comment on the command internal NI Automated Test Suites use in order to run system tests; and what it would take to change it to use (2) ?

adityabhat7, 2020-02-19:
In internal NI Automated Test Suites we have created python file to run system tests and we are calling that python file at the location of tox.ini file. We need to change the path to generated\niswitch and need to modify the python file to add '-c tox-system_tests.ini'.

PR #1269 (2020-03-02): Use per driver tox for testsuites
- [X] This contribution adheres to [CONTRIBUTING.md](https://github.com/ni/nimi-python/blob/master/CONTRIBUTING.md).
- [X] ~~I've updated [CHANGELOG.md](https://github.com/ni/nimi-python/blob/master/CHANGELOG.md) if applicable.~~
- [X] ~~I've added tests applicable for this pull request~~
### What does this Pull Request accomplish?
* Removes per driver system tests from `tox.ini` and `tox-travis.ini`
* `run_python_system_tests.py` modified to run test from the per driver tox.ini file
### List issues fixed by this Pull Request below, if any.

* Fix #1266 

###  [... cut]
closed 2020-03-02 by marcoskirsch