ni/nimi-python #1273: In nidigital, History RAM system tests cannot be run on simulated session

sbethur, 2020-02-13:
### Description of issue

nidigital python API contains fancy HRAM method. Digital driver does not simulate HRAM functions in the C API "well enough", to allow testing of the fancy HRAM method. During development of fancy HRAM method, the system tests were run on real h/w. But these tests are disabled on the nimi-bot runs.

We need to re-enable the tests by doing one of the following:

1. Mock the C entry points
2. Improve the simulation in C API
3. Use real h/w to run tests in CI
4. Something else

marcoskirsch, 2020-03-20:
The fact that simulation is deficient should be considered a driver bug. Please report that in the internal bug tracking system and post the number here.

I wouldn't mock the C API (in other words the _Library class). Through `nifake` we validate that the logic from the generator is correct. If you want to test the "fancy" handcoded function then mocking the private `nidigital.Session` methods it calls into is the right thing to do.

sbethur, 2020-03-20:
Created a bug in internal bug tracking system, number is 992370.

marcoskirsch, 2020-04-10:
This bug is about running the tests in simulation. It was closed by covering this code via unit testing. I think that's the workaround but this bug should remain open until driver bug 992370 is fixed and then these unit tests replaced with regular system tests.

PR #1395 (2020-04-09): Adding nidigital hram unit tests for site and pin behavior
- [x] This contribution adheres to [CONTRIBUTING.md](https://github.com/ni/nimi-python/blob/master/CONTRIBUTING.md).

- [x] ~~I've updated [CHANGELOG.md](https://github.com/ni/nimi-python/blob/master/CHANGELOG.md) if applicable.~~

- [x] I've added tests applicable for this pull request

### What does this Pull Request accomplish?

* Adds nidigital unit test `test_fetch_history_ram_cycle_information_pin_list`
* Adds nidigital unit test `test_fetch_history_ram_cycle_information_site_n`

### List issues fixed by this Pull Request below,  [... cut]