ni/nimi-python #1275: Inconsistent results from codecov

texasaggie97, 2020-02-14:
### Description of issue
We are getting inconsistent result reported by codecov into the PRs. For example, #1251 shows a coverage _decrease_ of 1.44% even though the only thing this PR does is add and remove blank lines in the .coverage.yml file.

I have started a [discussion](https://community.codecov.io/t/invalid-links-added-to-pr-on-github-plus-inconsistent-results/979/4) in the codecov support forums, but there has been no response.

thomasrockhu, 2020-05-15:
Hi @texasaggie97, Tom from Codecov here. I responded to you in the community thread, but would you be able to push a new commit on top of https://github.com/ni/nimi-python/pull/1251. Our systems have been upgraded in how we pick base and parent commits to compare against, and that might fix the issue. If not, I can go in manually and compare to see why this is happening for you.

sbethur, 2020-05-19:
Hi @thomasrockhu, master has drifted a lot compared to the state in #1251. I created another PR similar to that one - https://github.com/ni/nimi-python/pull/1457#issuecomment-630997122. Looks like the changes on your end has fixed the inconsistency problem.

Coverage report from some other recent PRs also look good - #1451, #1452, #1457 - no unexpected changes in coverage.